Klipper3d/klipper #462: CXY-V.2-0508

Teslastellar, 2018-07-06:
Hi,\
First of all, thanks a lot for such a great firmware. I finally managed to flash it and get it working on a Tronxy X8 and it works much better than Repetier or Marlin.\
I was wondering if we could add the config file and the flashing procedure for the board to the project config directory. The board is a Melzi-style board with Sanguino pinout (CXY-V.2-0508) and is used on a lot of printers from Anet and Tronxy and other Chinese manufacturers, so I thought it would help others with similar printers to start using Klipper.

jakep82, 2018-07-06:
Did you look at the generic-melzi.cfg file?  Not sure if it's the same as your board, but it includes a link to flashing instructions.

Teslastellar, 2018-07-06:
Yes I did. The pinout is slightly different. The only thing regarding flashing other than the bootloader was that avrdude flashed it only with 115200 baudrate, not slower. Took me a while to figure it out.
I left comments in the config file I made documenting where I got the bootloader and how I flashed it as well as how each setting can be found in Marlin configs. Just thought it would make other people's life easier if the config was included in the release. The printer's name, as mentioned, is Tronxy X8 but the config would work with other Sanguino boards as well.
I'm new to GitHub so I have no idea how I can contribute, but if it is of interest, I can copy paste the whole thing here or if there's a formal way of contributing to the project, I'd be grateful if someone could let me know.

jakep82, 2018-07-06:
Create a pull request with your new config file.  Details are at the link.

[https://github.com/KevinOConnor/klipper/blob/master/docs/CONTRIBUTING.md](url)

KevinOConnor, 2018-07-06:
As mentioned, you can submit the config via github pull request.  For a simple config file, it's also fine to cut-and-paste it here.

Just so I understand, this is a config that is different from config/generic-melzi.cfg?  Are you also submitting a Tronxy X8 config - or is it just the one config for both?

Thanks,
-Kevin

Teslastellar, 2018-07-07:
Thank you both for your quick responses :)

The config is for Tronxy X8 as I have tested it on that specific printer. The board looks different from Melzi but as far as I know it's based on Melzi and Sanguino boards. The pins for enabling z-axis steppers and the heatbed for example are different from the generic Melzi and the geometry, display, etc are all based on Tronxy X8.
I'll submit the file when I get on the computer.

Thanks again for the help and for all the effort to make such a powerful and versatile firmware.

Teslastellar, 2018-07-08:
Hi again,
Submitted via pull request, however, getting syntax errors from Travis. It was in comments so I removed two lines but still the same error on the same line.
Could someone please have a look?
Thank you,
Seyed

KevinOConnor, 2018-07-09:
There's a whitespace error in the config file (remove the trailing spaces at the end of line 100 - it currently reads as "lcd_type: st7920 ").

-Kevin

Teslastellar, 2018-07-11:
Apologize for the late response Kevin and thank you so much for letting me know what the error was, fixing it and adding the file.
I close this topic.
Keep up the good work.

-Seyed

closed 2018-07-11 by Teslastellar