Klipper3d/klipper #463: Heated Bed

Maximuscr31, 2018-07-06:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2171494/klippy.log)
I have finally got everything working well with klippy except for my heated bed. I have no bed option in the picture. When I first setup klippy I was not using a heated bed but have since gotten another and connected everything to ramps. Is there a setting I must enable to make it recognize the heated bed and use it? I can reset it all up again if necessary but prefer to fix it if able. 
[image]

mmone, 2018-07-06:
There is a typo in the config "heater_bed" instead of heated_bed

mmone, 2018-07-06:
Do you have "Heated Bed" activated in your OctoPrint printer profile?

Maximuscr31, 2018-07-06:
I did not. That took care of it. Thank you so much.

closed 2018-07-06 by Maximuscr31