Klipper3d/klipper #466: Set different Travel acceleration and printing acceleration

StefanMeGit, 2018-07-08:
Is there any possibility to set travel acceleration and printing acceleration differently?

KevinOConnor, 2018-07-10:
That isn't supported.

It is my opinion that adding support is not worthwhile.  Even for non-extruding moves the nozzle is often still in contact with the print (it is common to have XY moves on the same level as extruding moves).  Using a higher acceleration when the nozzle is in contact with the print does not make sense to me.  It might make sense to use a higher acceleration when the nozzle is not in contact with the print, but the firmware can't readily determine that, such moves are relatively rare (thus unlikely to notably impact print time), and the slicer would be in a better position to implement that (via M204).

-Kevin

hg42, 2018-07-10:
in fact I know Marlin and Smoothieware behave different in this case.
Because of this, slicers either allow to configure different flavors of gcode or always set the F-parameter and don't rely on different speed settings.

StefanMeGit, 2018-07-10:
I think its an important feature for a nice retract and for less oozing while traveling. If u can "run away" very quickly after a retract special on printing situations where u go down to a very small acceleration it would be helpful. 

After I made the outer perimeter with only acc 1500, I want to travel away from the end point as fast as possible with max acc to decrease oozing or other effects. Its a basic function in Marlin and Smoothieware. http://marlinfw.org/docs/gcode/M204.html

I would love to see this in Klipper!

KevinOConnor, 2018-07-17:
FYI, the M204 command is supported (at least with the S parameter).

closed 2018-07-17 by KevinOConnor