Klipper3d/klipper #467: Setting extruder position using G92 to a large value gives an error.

oderwat, 2018-07-08:
I just ran into a problem where I needed to set the extruder position to a large value for continuing a print where the gcode uses absolute extrusion lengths (Curas default).

When I do something like this:

```
G28
G92 E3861
G1 X10 Y100 E3862
```

I get an error that the extrusion is more than what is allowed by `max_extrude_only_distance` (with a negative value).

I think this is a bug and klipper should not care about what one sets as position for the extruder.

KevinOConnor, 2018-07-10:
In order to investigate an issue we need the Klipper log file.  See:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

oderwat, 2018-07-10:
I think I figured it out. I guess it had to do with switching relative and absolute modes. Can't reproduce it now.

closed 2018-07-10 by oderwat