Klipper3d/klipper #470: Support STM32 Variants

sniperlucian, 2018-07-09:
I am happen to have a STM32F0xx and a STM32F4xx board laying around, but no STM32F1xx.

For AVR you support nicely different variants. I think that would also make sense for the STM32 family. 

Having experience with them, I would also consider to help porting. Could you document you workflow (which HAL libs etc.) ? So its easier to get starting.

yschroeder, 2018-07-09:
@grigorig might be able to help here.

grigorig, 2018-07-09:
The differences between the STM32 families are bigger compared to AVR. There are various different (and incompatible) CPU cores in use and different peripherals, to some degree. The low-level HAL that I am using in Klipper only offers a bit of abstraction, which probably isn't really enough to support a wide variety of MCUs. I haven't actually checked how much work it would be to support F0 or F4 series, though.

My guess is that it would be the easiest to copy and adapt the STM32F1 port based on the low-level HAL for another STM32 series and then extract out some common code as you go. It's probably quite feasible to share most of the GPIO, ADC and SPI code.

dragonnn, 2018-07-10:
I am not an expert, much more a beginner in STM-s, now I am working on a project using libopencm3 exactly. First impressions - I really like it and I even get full debugging on Linux with Visual Studio Code and Cortex Debug plugin.

grigorig, 2018-07-10:
libopencm3 doesn't provide suitable generic abstraction either. It's not even possible in many cases: the different MCU families have different features that sometimes work in different ways, so they can't be used the same way with just a thin abstraction layer.

yschroeder, 2018-07-10:
What about using Cube MX to generate the HALs? Maybe they are similar enough for easy implementation of all platforms?

Or you could take a look at RIOT OS. They also support all the different STM chips. They have some `stm_common` code and then extra code for all the different chips:
https://github.com/RIOT-OS/RIOT/tree/master/cpu

KevinOConnor, 2018-07-10:
FYI, there was a long discussion on possible HAL and RTOS options in issue #137.

-Kevin

sniperlucian, 2018-07-10:
I see - it's general awareness - not even including the STM'S but for all micros. 

So what would be the best way for the STM's now - new main device - or a variants?  Also in terms of maintenance ...

ysoldak, 2018-07-11:
Also want this. STM32F0xx specifically.
Linking official F1<->F0 migration doc from STM32 if that's any help.
https://www.st.com/content/ccc/resource/technical/document/application_note/19/dc/60/19/9b/8b/45/ce/DM00052530.pdf/files/DM00052530.pdf/jcr:content/translations/en.DM00052530.pdf

dim08, 2018-07-11:
In my opinion support for STM32F407VET6 might be useful, some inexpensive boards are available. But actually instead porting each MCU wouldn't be easier to port a RTOS like RT-Thread (https://github.com/RT-Thread/rt-thread) that already supports quite a few architectures ?

sniperlucian, 2018-07-12:
actually the functionality of klipper on the board itself is very limited and low level. an RTOS wouln't bring any benefit. The most critical low level code is the stepper timing - which probably is the most difficult to port.  

(having said this - I am planning to add much more sensors to the board - and an RTOS would help to collect all those data (i2c thread - uart thread etc...))

At least  the HAL from  http://www.chibios.org/  claims:

_**HAL includes a rich set of features making writing application a simpler task. The user has no necessity to understand all the inner working of MCU. 

-SPI, abstraction of an SPI master with callback capability.
-UART, abstraction of an UART unit with callback capability and unbuffered handling.**_

Also they seem to have nice license: 

**_This component is only released only under Apache License 2.0 and is 100% free also for commercial use._** 

So it would *only* come down to implementing the timer.

ysoldak, 2018-07-13:
Some guy forked Marlin and made a generic STM32 HAL, check out here.
https://github.com/xC0000005/Marlin/commits/genericHal/Marlin/src

Hope that helps that brave soul who takes the task of supporting different flavours of STM32 in klipper.

KevinOConnor, 2018-07-30:
It's fine if someone wants to implement the port and try it out.  Let us know.

closed 2018-07-30 by KevinOConnor