Klipper3d/klipper #471: MCU 'mcu' shutdown: Move queue empty

bdanmcm, 2018-07-09:
4 Hours into a print at 99% done.  Printer stops, turns off heaters and shows error:
MCU 'mcu' shutdown: Move queue empty

Printing on stock anet a8 board @60mm/s.  So far this is the only print this has happened on but it is the longest print attempted using klippy.

Attached is the log

Thanks
Brendan

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2176141/klippy.zip)

KevinOConnor, 2018-07-09:
Unfortunately, you've hit the same bug described in issue #392.

[image]

bdanmcm, 2018-07-09:
Thanks kevin, I just removed the display to see if there was an issue with that.  I will try today with the CLOCK_MONOTONIC_RAW change on a longer print and update tonight/tommorrow.

bdanmcm, 2018-07-10:
Just tested on an 11 hour print.  Printed perfect, no issues after changing to CLOCK_MONOTONIC_RAW.  Closing issue...

closed 2018-07-10 by bdanmcm