Klipper3d/klipper #475: Output_pin software PWM static value error

KostteT, 2018-07-11:
Good day, users and developers this great project.
Why klipper does not want to accept cfg with the following parameters:
`[output_pin fanboard_pin]
pin: ar7
pwm: true
static_value: 0.5`
answer log:
`start and shutdown values must be 0.0 or 1.0 on soft pwm Once the underlying issue is corrected, use the "RESTART" command to reload the config and restart the host software. Printer is halted`
In this case, this function differs little from static_digital_output

I understand what can be done in another way 
`value: .0` and send `SET_PIN PIN=fanboard_pin VALUE=.5` Before the start of printing
But I want to do it correctly and automatically. In that case, maybe there is a way to link the cooling with the engines on automatically? And make a delay of 20-30-60 seconds after stopping? This option is even better than turning on always
best regards from Siberia )

KevinOConnor, 2018-07-12:
Handling startup and shutdown values for software based PWM is tricky and thus has not been coded.

Most users with a cooling fan use a [heater_fan] section - see config/example-extras.cfg.

-Kevin

KostteT, 2018-07-13:
the simplest solution: https://www.thingiverse.com/thing:1274733
Only change is used: diaganal rod (arm_length) for ich axis

closed 2018-07-22 by KevinOConnor