Klipper3d/klipper #476: How to adjust the print sizes X/Y delta

KostteT, 2018-07-11:
How to adjust the size of the print, did not find no one parameter to adjust, is it really only the manually correct length of the rods and the move to a step? Or give a link please, where can I read about this
Calibration, of course, did not go well, as in marlin with adj rod and adj tower, but at first glance, printing the first layer is even better than marlin without adj rod and adj tower, however, the size print not good (
Delta, glass 6mm on the table
Thank you in advance

KevinOConnor, 2018-07-12:
On a delta printer, one can use the DELTA_CALIBRATE command to perform basic delta calibration.  This calibration should get the bed level, but will not calibrate part dimensions.  In order to do that, I think one would need to print out a calibration object and then follow one of the delta calibration guides available online.

In the future, please be sure to attach the klipper log file on any requests for assistance.

-Kevin

KostteT, 2018-07-13:
the simplest solution: https://www.thingiverse.com/thing:1274733
Only change is used: diaganal rod (arm_length) for ich axis
Do not confuse the values delta_radius and horizontal radius - latter is best measured manually.

ysoldak, 2018-07-14:
I like this one better -- also helps fix tower angles
https://www.thingiverse.com/thing:745523

KevinOConnor, 2018-09-06:
FYI - I've put together a development branch with enhanced delta calibration.  See #615 for details.

-Kevin

closed 2018-09-09 by KevinOConnor