Klipper3d/klipper #478: Reconnect after Power Off

sniperlucian, 2018-07-12:
To prevent burning of my home I switch the printer off when I am not home.

However - after switching on - klipper not automatically reconnects after power on of the boards. So I usually need to:

- switch power on of printer
- need to restart klipper (etc/init.d/klipper restart)
- need to unplug the USB / replug USB
- need to restart klipper again

- finally klipper starts to log the actual sensor data

btw- I am using two boards - RAMPS and ANET A8.

Since klipper runs as deamon - I would expect it to reconnect as soon the printer comes online again, or is this normal/wanted behavior?

KevinOConnor, 2018-07-14:
> need to restart klipper (etc/init.d/klipper restart)

Use the RESTART command instead.

> need to unplug the USB / replug USB

Use a different serial config option to avoid this - see the FAQ: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

> need to restart klipper again

Just the one RESTART command should be all that is necessary.

> Since klipper runs as deamon - I would expect it to reconnect as soon the printer comes online again, or is this normal/wanted behavior?

It is intentional so as to prevent an error from escalating into a much worse problem.  In the event that a micro-controller disconnects we want the user to acknowledge it (via a RESTART or similar command) before continuing.

-Kevin

sniperlucian, 2018-07-20:
This is not working ;(  - Klipper still refuses to restart the controller board.
 
I have two boards. One keeps running cause it gets power via USB - the other switches off. Could that be the reason that I can not restart it with "RESTART_FIRMWARE" ;  "RESTART" ? Maybe run away time counter?

KevinOConnor, 2018-07-22:
If something isn't working correctly, you'll need to attach the full klipper log in order for us to assist.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

sniperlucian, 2018-07-29:
Looks like that I didn't use a # in front of commands - which is necessary for repetierhost  to pass it trough to klipper.  Its however not necessary for just FIRMWARE, but it cuts anything after '_' there fore needs the #.

closed 2018-07-29 by sniperlucian