Klipper3d/klipper #479: Mover after Kill Print

sniperlucian, 2018-07-12:
Moving without Endstop has been discussed here several times. 

However - I needed to kill several prints now - and I really needed to move Z,  and since I have some printed part under the extruder here is NO way to home Z. And its not easily done by hand as with XY.

Would it be possible to keep the state of endstop after killing a print?  Since the home is usually still valid and killing is an manual process too, the operator is aware of a potential problem and it should be a save option.

jakep82, 2018-07-12:
Octoprint disables motors by default when a print is canceled.  If you change this setting so motors aren't disabled, you can move Z without the need to re-home.

scarecrow2k6, 2018-07-13:
@jakep82 Tried what you suggested. From Octoprint, cant turn on the motor after cancelling a print.

sniperlucian, 2018-07-13:
I am using klipper with repetierhost desktop host. works without problmes with repetier firmware too.

jakep82, 2018-07-13:
See the link for more info.  The default gcode that Octoprint runs after a print is canceled includes disabling the motors (M84).  If you delete this line in the Octoprint settings you can move the head without re-homing.  You can also include other things like parking the head at X0Y0.  This is not a Klipper bug, just a bad default setting in Octoprint.

https://github.com/foosel/OctoPrint/issues/1399

closed 2018-07-22 by KevinOConnor