Klipper3d/klipper #480: Separate MCU For BLTouch

r3donnx, 2018-07-13:
Hi! 
I see, that Klipper can work with multiple MCU's. My printer (Wanhao Duplicator i3) uses Melzi board and it has very limited number of I/Os, so it is quite challenging to connect BLTouch sensor to it without loosing some other functionality. I wonder if it is possible to use Klipper and separate Arduino with BLtouch connected to it for bed leveling? :)

jakep82, 2018-07-13:
I'm sure Kevin can give a more precise answer, but to make this work I believe you would need the Z stepper to run on the same MCU as the BLTouch.  It's not possible to run endstops/Z-Probes on a different MCU than the one controlling the stepper.

r3donnx, 2018-07-16:
Thanks @jakep82. That makes sense. I guess I'll have to sacrifice LCD for BLTouch until the board upgrade.

closed 2018-07-16 by r3donnx