Klipper3d/klipper #481: MCU 'mcu' shutdown: ADC out of range  -- But no sensors is high or low??

schwab, 2018-07-14:
MCU shutdown keeps happening about 1/2 through a benchy boat print, but both temperatures (bed and hothead) are normal.  What else can be causing this? This is such a difficult one to troubleshoot. It would be nice if the "bad" values where displayed in the log along with the errors.  How does one find the actual cause of this error?

[klippy.log](https://github.com/KevinOConnor/klipper/files/2194510/klippy.log)

KevinOConnor, 2018-07-14:
That's certainly odd.  You should update to the latest version of the code, as noise filtering on the adc range check was added just a few days ago.  It also improves adc debugging.  (See the FAQ for details on upgrading.)

-Kevin

schwab, 2018-07-14:
Thank you for the quick response.  With the newest software installed and firmware updated, I was quickly able to find extremely high values on the hot head thermistor (right there with the error message) the next time it failed. The wire for it was being pinched in the z-axis which explains why the print could work up to a point then fail.

closed 2018-07-17 by KevinOConnor