Klipper3d/klipper #482: Question about TMC2130 in HW SPI mode

Trontastic, 2018-07-14:
Hello guys!

I'm currently merging my printer from Marlin to Klipper and I have a question about the TMC2130 drivers.

I connected 5+3 cables to the board:
- SDO (shared between all drivers)
- SDI (also shared)
- SCK (also shared)
- X CS
- Y CS
- Z CS
- E0 CS
- E1 CS (used as Z2)

The problem is that I don't find any pin configuration options in Klipper for the SDO/SDI/SCK pins.

Can you please explain it why?
Are these not needed anymore or they have a default value?

Thanks!

fbitto, 2018-07-14:
I use an Arduino 2560 + RAMPS 1.4 and the SPI pins are hardwired. I think you can't reassign them.

Trontastic, 2018-07-14:
Oh okey, I just checked the source code and I see them they use the same pins as Marlin, so they will work the same without rewiring them to another ones.

I just missed this from the documentation.

Thanks :)

closed 2018-07-14 by Trontastic

CHILLYSMOKES, 2018-07-14:
You can reassign them. Check the "examples-extras.cfg" file. Search TMC2130 and a couple sections down from that check the LCD display example on the call out. The LCD shares all the same SPI pins but for each motors cs pin.

Tahutipai, 2018-09-25:
If I'm reading this correctly, you can only set the spi-clock pin on SPI, not the MOSI or MISO. Can you confirm?