Klipper3d/klipper #484: Following PEP8

mamoit, 2018-07-16:
I've been trying to read through this project, and it seems that following [PEP8](https://www.python.org/dev/peps/pep-0008/) could do a lot in terms of readability.

oderwat, 2018-07-16:
You may want to do like me and let PyCharm re-format everything to make it more readable and allow inspection of the code more easily.

mamoit, 2018-07-16:
That's one option, but if you're planning on contributing, then there is a need to keep a consistent (readable) formating throughout the project, since a patch developed on a re-formated codebase may not apply correctly on the repo.

oderwat, 2018-07-16:
I know...

KevinOConnor, 2018-07-22:
I don't think github issues are a good place for discussing high-level topics like this, as they tend to never be "complete" and thus stay open forever.  The mailing list or IRC might be a better place (see https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md for details.)

Further, I don't think a discussion on coding style will be productive.  I fear it will just lead to "bike shedding".

mamoit, 2018-07-23:
Fair enough.

closed 2018-07-30 by KevinOConnor