Klipper3d/klipper #485: Running Klipper as a Linux real-time process ?

dim08, 2018-07-17:
I am interested about running Klipper as a Linux real-time process. In other words using CPUs native GPIOs, SPI, I2C ports etc. Noticed there is some support in src/linux/, but no much info how this can be used. Would you be able to provide additional information ?

KevinOConnor, 2018-07-18:
The Klipper micro-controller code is capable of running as a Linux process with real-time scheduling.  This code (in src/linux/) is used with the Replicape to provide some basic ADC, I2C, and SPI capabilities that that board requires.  Other interfaces (such as GPIO) have not been implemented.  It would be possible to do further extend the code to add additional capabilities.  However, I suspect controlling a stepper motor at high speeds would be impractical via a Linux real-time process.

-Kevin

dim08, 2018-07-18:
I tried compiling it on x86 platform and it didn’t generate bin file, just elf. Also when tried:
objcopy -O binary klipper.elf klipper.bin
got:
-bash: ./klipper.bin: cannot execute binary file: Exec format error 

Regarding the control of step motors. Some Internet post state that data transfer over 25 MHz via bitbang on RPi might be feasible. If custom Linux kernel driver is written just for the step motors control ?

dim08, 2018-07-19:
@FHeilmann, I did some profiling, the Klipper and Repetier-Server processor usage is not much. As well as their processor usage probably can be caped by some utilities ?

i.MX233 offers about 500 DMIPS. BCM2837 (RPi 3) offers 4x ~2760 DMIPS, totaling 11040 DMIPS which seem like a total waste just for generating step motor signals. I’ve done some comparison for running Klipper on bare-metal, not clamming for accuracy, see the image bellow. The expected step rates (red) were calculated based on the info from (Features.md).

In my opinion, there suppose to be some way to achieve near real-time execution in Linux. Probably it requires some additional research.

[image]

FHeilmann, 2018-07-24:
It's not that simple unfortunately. 

Feel free to try but prepare yourself for nasty concurrency effects once you move to speeds which require tight timing guarantees.

closed 2018-07-30 by KevinOConnor