Klipper3d/klipper #486: TCM2130 SPI ISSUE WITH RE-ARM

CHILLYSMOKES, 2018-07-17:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2202726/klippy.zip)

Re-Arm with Ramps 1.4
TMC2130 drivers x4
LCD working
BLTouch <<---------- does self check at start, light turns on, have not been able to test, yet

I am getting the following error in the terminal when I connect.

Recv: // Unknown command: spi_send
Recv: // This type of error is frequently caused by running an older
Recv: // version of the firmware on the micro-controller (fix by
Recv: // recompiling and flashing the firmware).
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Protocol error connecting to printer
Recv: !! Unknown command: spi_send

Status was good until I added the [tmc2130 stepper_x] section.  When it is removed there is no issue.  I have recompiled several times to no avail.  I have moved the X-CS pin around the board and no change.  In Marlin I was able to use LCD with TCM2130s by reassigning the x,y,z,e cs pins while the other SPI pins were shared with the LCD.  What am I missing?  Many thanks.

KevinOConnor, 2018-07-18:
Unfortunately, the SPI interface has not been implemented for the LPC176x.  I'll try to take a look at it over the next few days.

-Kevin

KevinOConnor, 2018-07-18:
I added basic SPI support to the LPC176x code.  You should grab the latest code and retry (see the FAQ for details on upgrading the software).  You will need to reflash the mirco-controller.

-Kevin

KevinOConnor, 2018-07-18:
Note that the st7920 does not use SPI (it uses a bizarre proprietary protocol) and it must not use the standard SPI pins.

-Kevin

CHILLYSMOKES, 2018-07-19:
THANKS!!!  My driver's work now so I am extremely happy. Still looking for an available CS pin, only need 1 now.  Had to trial and error it even after figuring out how to translate Ramps pins to Re-Arm. For anyone searching for answers use 3 pins on Jumper5, not the 4th on the bottom left, it interferes with BLTouch AND the LCD speaker whines.
@KevinOConnor - Note well taken... the LCD no longer functions. If you can point me in a direction I can brute force some simple stuff on my own. Like, is this a problem with figuring out SPI pins or is this a high level coding issue?? Is it possible to add a second MCU for the LCD? Does SPI work over multiple MCU? Could I use CS pins on the second MCU? Thanks for your time. You are appreciated.

AxMod3DPrint, 2018-07-19:
Just as a note with regards to pinout, there is a generic config for the
Re-Arm and the pinout is freely available on the Panucatt website.

'Eagles may soar, but Weasels don't get sucked into Jet engines'

KevinOConnor, 2018-07-19:
On Wed, Jul 18, 2018 at 06:40:43PM -0700, CHILLYSMOKES wrote:
> @KevinOConnor - Note well taken... the LCD no longer functions. If you can point me in a direction I can brute force some simple stuff on my own. Like, is this a problem with figuring out SPI pins or is this a high level coding issue??

Neither the st7920 hardware nor the software supports sharing the
st7920 pins with another device.

>Is it possible to add a second MCU for the LCD? Does SPI work over multiple MCU? Could I use CS pins on the second MCU? Thanks for your time. You are appreciated.

It is possible to connect an LCD to a second MCU.  It is not possible
to split the LCD pins across multiple MCUs though (all the pins for
the display must be on one MCU).

So, for example, you should be able to use an arduino nano or uno to
pick up a few more pins.

-Kevin

mmgarcia30, 2018-07-19:
> Yes but we don't plug stuff directly into the Re-Arm. Ramps lost 8 digital pins when upgrading to Re-Arm from Mega, ie- All the AUX 2 digital-only pins. Which were where all my CS pins were... If you could point me to a Ramps pinout with Re-Arm aliases that would actually help. I am using the generic config as the base for my config, too.

But you have all the data servo pins available to CS  or as digital pins in your Ramps board. Just need to define them  in klipper.

AxMod3DPrint, 2018-07-19:
Pinout for the re-arm is at https://bit.ly/2Lv8X0Z

'Eagles may soar, but Weasels don't get sucked into Jet engines'

CHILLYSMOKES, 2018-07-20:
Thanks for all the advice, guys.

@mmgarcia30- dunno why, but the servo Digi pins aren't working. BLTouch not working correctly, too. I may have translated the pins wrong, though I was able to translate everything else... I'll take another look. 

@AxMod3DPrint- I just saw another post about the LPC1768 having extremely noisy thermistor inputs and thought about firing up the soldering iron to bridge those buffered pins and use them. Your suggestion just solidified it. Thanks. The link you provided doesn't work, though. I'll try that header, too.

@KevinOConnor- thanks for clearing that up for me. Now I can weigh my options.

closed 2018-07-20 by CHILLYSMOKES