Klipper3d/klipper #487: Update Klipper to latest release or commit to master branch?

kevinmce, 2018-07-19:
Hi @KevinOConnor I wanted to ask something as it was  not obvious from the installation guide or FAQ.

When installing Klipper using the [guide ](https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md)do we get the latest release version V0.6 or the latest master branch version? I suspect its the latest commit to the master branch?

Also I guess the [update procedure](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software) in the FAQ will have the same result.

AxMod3DPrint, 2018-07-20:
You're better off pulling the master branch as a lot of improvements have been made since 0.6

kevinmce, 2018-07-20:
So what should i change in the upgrade procedure to get the master branch?

Or has "git clone https://github.com/KevinOConnor/klipper" already got the master branch?

oderwat, 2018-07-20:
tl;dr: You have the master.

As long as the remote repository has this as the default branch (where the remote repositories HEAD points to) and you don't change the branch with a `git checkout <branchname>` command after cloning the branch.

So if you want to be sure to be at master you can also do an: `git checkout master` or check if you are at master by `git status`. Your current branch also has a '*' when you do `git branch`, which lists all the available branches.

kevinmce, 2018-07-20:
Ahh nice. I think that explains everything i wanted to know. 

Much appreciated!

closed 2018-07-20 by kevinmce