Klipper3d/klipper #489: Import settings from Marlin?

kevinmce, 2018-07-20:
This is more of a feature request than an issue, but after converting from marlin myself this week I noticed that the job of configuring klipper was made way easier by being familiar with the same settings in marlin.

What would be great for the less experienced persons in the community is if there was a way to import your general configuration from marlin? 
This might include:
Stepper Pins
Stepper direction
Step distance
kinematic type
build volume
number of hotends/beds
thermistor types

I would say 90% could be directly taken from a previous marlin config. 
It certainly would open up the firmware to be more accessible to new users.

gdachs, 2018-07-21:
A small standalone C program that would include the config header file from marlin during its build could easily spit out the klipper config.

KevinOConnor, 2018-07-22:
I suspect this would be challenging to implement in practice, but if someone wants to try then that's great.  It's not something I plan to work on in the near future.

Cheers,
-Kevin

KevinOConnor, 2018-07-30:
I'm going to close this for now - we can reopen should someone start development on it.

-Kevin

closed 2018-07-30 by KevinOConnor