Klipper3d/klipper #491: Tool selection is not stored between reset.

kevinmce, 2018-07-21:
So, jist setup my dual hotend system with docking. 

So using the activate/deactivate scripts the dock and undock work correctly so long as i always reset with T0 on the carriage. But if T1 is attached during reset it thinks T0 is attached after. Resulting in a crash after restart

kevinmce, 2018-07-21:
Crap sorry. My mistake, let me do more testing but it looks like my hotend exchange crash was a once off. 

Been testing with resets since and its been bullet proof so far. Will report if i find anything different. 

For now are some hotend exchanges for your viewing pleasure. :)

https://youtu.be/Nj3CPNZZzTs

KevinOConnor, 2018-07-22:
Interesting.  In the future, should you run into a problem, you'll need to attach the full klipper log file in order for us to assist.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

kevinmce, 2018-07-22:
Actually i have confirmed the issue of lost tool selection on reset while testing the extruder offset command. 

I had started a print with T0, it failed during the tool change back to T0 due to incorrect offset during the tool deactivate.
Afterwards i performed a reset, leaving T1 on the carriage, i then homed XY and issued a T1 command to see if it would try to change the tool or just sit there. It proceeded to try dock T0 and load T1 when it should not do anything. 

I'm not sure what the best way to handle this is. it is maybe best to enforce a T0 start always due to people will likely start with that tool after working on their printer. But it seems a little fiddly that every time i get a failure on T1 i risk forgetting to manually load T0 and crash the hotend. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/2217146/klippy.log)

KevinOConnor, 2018-07-23:
I'm not sure what you mean by "reset".  If you mean after issuing a RESTART command then that is to be expected.  Unless the toolhead has a sensor capable of detecting which head is loaded (and code is written to read that sensor) how would the software know which head is loaded upon startup?

If there is some sequence of commands that is always safe to run that will unload the head (or similarly, some sequence of commands that is always safe to run prior to a restart), then it may be possible to change the host software to do that on each startup and/or restart.

You may also find it useful to define a gcode_macro (see config/example-extras.cfg) for the load and unload sequences, and call that macro from activate/deactivate_gcode.  That way you can also manually run the macros should you need to.

-Kevin

kevinmce, 2018-07-23:
Sorry, I meant the RESTART command, that was what i used. 

But i think i came to the same conclusion, the best way to hanfle a restart is to assume T0 is active. 

I dont think i have a way to easily unload all heads and load a specific head. It would require some clever docking mechanisum. 

The macro seems like a nice idea. At least then i could correct a tool mismatch without handling the hotend.

closed 2018-07-25 by kevinmce