Klipper3d/klipper #493: Possible to set extruder offset?

kevinmce, 2018-07-21:
As far as i can see there is no way to set an offset between hotends. 

Not an important feature but might be nice to have this controlled in the firmware.

KevinOConnor, 2018-07-22:
I'm not really sure what you mean by "offset between hotends".  However, you should look at the SET_GCODE_OFFSET command, and look at the activate_gcode and deactivate_gcode descriptions for extra extruders in config/example-extras.cfg.

-Kevin

kevinmce, 2018-07-22:
Ahh thats exactly what i was looking for. 

I thought it would be part of the extruder config. But as part of the activate/deactivate gcode is good too. 

Ill leave a link to the[ extended gcode section ](https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#extended-g-code-commands
)for anyone searching the same info in the future.

closed 2018-07-25 by kevinmce