Klipper3d/klipper #494: BLTOUCH z_offset / z position_endstop question

CHILLYSMOKES, 2018-07-21:
I ran bed_tilt_calibrate and it gives me values and whatnot, but my BLTouch triggers when the nozzle is 1.6mm from the bed.  So after the calibration the print fails because the nozzle is way too far from the bed.  Is there an argument I can put in that will subtract 1.6mm from every G-code z-move??  I have changed the z-offset in slic3r but it isn't working (possibly because the z_offset in the config is overriding it??).   What am I missing?  It's got to be simple because no one else is complaining. Thanks!!

oderwat, 2018-07-21:
You want to define an offset in the `[probe]` section. You want the same offset in the in the stepper_z section if you use the BLTouch for homing the z axis.

```
[probe]
...
z_offset: 1.0
...
[stepper_z]
endstop_pin: probe:z_virtual_endstop
position_endstop: 1.0
...
```

CHILLYSMOKES, 2018-07-21:
Thanks for the reply, oderwat.
I had done that already.  I found the issue though.  I have a nozzle wipe area I have the extruder pass over right before the print starts and the START GCODE has this block:

G1 F2000 X20 Y220 Z5      ; PARK
M400       ;  DO ABOVE FIRST
G1 F5000 X20 Y210 Z0     ; AUTO CLEAN NOZZLE BLOCK
G1 F5000 X75 Y220 Z0       ;  BAD COMMAND
G1 F5000 Z5        ; 

It didn't like the fourth command because that would move the nozzle to a "negative" Z-space based on the calibration (like -0.05 or something) so it skipped the entire block and kept Z=5 when it started the print.  Anyway, I changed the START GCODE to this:

G1 F2000 X20 Y220 Z5      ; PARK
M400       ;  DO ABOVE FIRST
G1 F5000 X20 Y210 Z0     ; AUTO CLEAN NOZZLE BLOCK
G1 F5000 X75 Y220 Z1       ;  GOOD COMMAND
G1 F5000 Z5        ; 

Z=1 to do the wipe and its printing perfectly now.  Your reply made me look in a different direction and I found it.   Thanks, again.

closed 2018-07-21 by CHILLYSMOKES