Klipper3d/klipper #495: Possible speed improvement for STM32F1 port ?

dim08, 2018-07-21:
I’ve been doing some speed comparison of MCUs architecture implementations posted in [#485](https://github.com/KevinOConnor/klipper/issues/485). Apparently the ARM ports are not as fast as AVR version in terms of Kilo Steps / DMIPS.

Looked at:

stm32f1xx_ll_gpio.h:
__STATIC_INLINE void LL_GPIO_ResetOutputPin(GPIO_TypeDef *GPIOx, uint32_t PinMask)
{
  WRITE_REG(GPIOx->BRR, (PinMask >> 8 ) & 0x0000FFFFU);
}

So far I was not able to get Klipper to generate proper assembler listing file to examine the code. But one possible solution: defining of another PinMask array to avoid the shifting and AND operation ?

KevinOConnor, 2018-07-22:
I view the generated arm assembler with the following command: `arm-none-eabi-objdump -ldr out/klipper.elf | less`

It is possible a minor speed improvement would be noticeable by avoiding the stm "low level" library in the gpio toggle function (the other gpio functions are not performance sensitive).  Only way to know is to test it.  Also, @grigorig might have some comments.

-Kevin

grigorig, 2018-07-22:
Well, AVR is more efficient mostly because it doesn't need to schedule unstepping.

Optimising the GPIO code might or might not do anything (I haven't checked the assembly), but I'd expect the improvement to be very minor. Most of the overhead is in the actual scheduling code and interrupt handling.

dim08, 2018-07-23:
Thanks for the info. If I understand correctly the unstepping is done to ensure the timing requirements of a step motor driver min/max step time. For DRV8825 – 1.9 uS, for newer step drivers – TMC2208 – 100 nS.
Instead of scheduling unstep events in case of faster MCUs wouldn't be faster to add just some adjustable delay to meet the timing requirements of step motor drivers used ?

KevinOConnor, 2018-07-23:
> Instead of scheduling unstep events in case of faster MCUs wouldn't be faster to add just some adjustable delay to meet the timing requirements of step motor drivers used ?

The mcu C code has been arranged so that stepping code (stepper.c:stepper_event() ) and scheduling code (timer_irq.c:timer_dispatch_many() ) both get inlined into the hardware timer IRQ handler.  The scheduling code already implements a busy wait for events that are scheduled in the near future.  So, it is possible to busy wait in the stepping code, but that isn't a huge gain over busy waiting in the scheduler.  Busy waiting in the scheduler has the advantage of allowing other timers (eg, steps on other steppers) from proceeding while waiting for an "unstep" event.

When I first wrote the stepping code that uses rescheduling, I tried a few different methods and ran some benchmarks on them.  In general, I found that with some tricks I could significantly increase the single stepper benchmark, but those tricks had little (or a negative) impact on the 3 simultaneous stepping benchmark.  Since the multiple active stepper benchmark is much more interesting in practice, I felt the simple, rescheduling, stepper code was the best approach.

This is also why the code uses a static 2us step pulse duration.  I found that using smaller times had no improvement on the 3 simultaneous stepper benchmark - so the additional complexity of defining variable length delays was unnecessary.

-Kevin

grigorig, 2018-07-25:
FWIW I had the idea of using timed DMA writes for the unstepping, but I never implemented it, mostly because Klipper is already plenty fast enough for my purposes.

closed 2018-07-30 by KevinOConnor