Klipper3d/klipper #496: Some questions about the SPI implementation

FHeilmann, 2018-07-24:
As you are probably aware I am working on a TMC2660 extra for klipper.
In contrast to the TMC2130, the 2660 want to reply to every SPI packet sent, so trying to do spi_send results in the dcommunication locking up and not accepting any more commands.

I since switched to SPI transfer, however, this generates a lot of logspam during the setup phase (where I am transferring data to the TMC2660 using the following code adapted from the tmc2130 extra:

```python
    def add_config_cmd(self, val):
        self.mcu.add_config_cmd("spi_transfer oid=%d data=%06x" % (
            self.oid, val & 0xffffff))
```
since the response is not handled by this code, it is dumped to the log. Is there any way to avoid this? i.e. add the send_with_response as a config_cmd?

KevinOConnor, 2018-07-24:
The SPI protocol always transmits and receives simultaneously.  The only difference between spi_send and spi_transfer is that the former discards the response data in the micro-controller code.  If you were using the sam3 spi code as a reference, be sure to apply the equivalent of commit 1eb4160, as that fixed a bug preventing spi_send from working correctly on the sam3.

To your question about receiving spi_response messages - one would use mcu.register_msg(my_callback, 'spi_response', my_oid) to register a callback for each spi_response message (on that particular oid).  If a callback is registered, it wont spam the log.  However, as above, this shouldn't be necessary for basic configuration of the tmc chips.  Note, if you do register a callback with the register_msg() mechanism, the callback is invoked from a background thread.

-Kevin

FHeilmann, 2018-07-24:
Thanks for the reply, I am specifically interested in combining the "add_config_cmd" with the "send_with_response" functionality of klipper. I saw how to do it outside of the configuration and initialization, but I couldn't figure out what the API is to have config_cmds use the response as well

closed 2018-07-24 by FHeilmann

KevinOConnor, 2018-07-24:
On Tue, Jul 24, 2018 at 01:47:38PM -0700, Florian Heilmann wrote:
> Thanks for the reply, I am specifically interested in combining the "add_config_cmd" with the "send_with_response" functionality of klipper. I saw how to do it outside of the configuration and initialization, but I couldn't figure out what the API is to have config_cmds use the response as well

The config_cmds are, in general, built in advance of connecting to the
micro-controller.  It's the list of commands that would initialize the
micro-controller.  When the Klipper host does connect to the
micro-controller, it compares the crc of the local config commands
with the crc of the config last sent to the micro-controller, and uses
that to determine if the mcu needs to be reset or not.  So, waiting
for a response from a config command doesn't really make sense.

I suspect the closest equivalent to what you're asking would be to
delay sending commands until the printer_state('ready') callback is
invoked, and then using send_with_response() at that point.

-Kevin