Klipper3d/klipper #497: Klipper restart needed for TMC2130 after power off

AxMod3DPrint, 2018-07-25:
Wierd one, and probably an edge case. Looking at the log it seems the clock is losing sync. Running commit acefe26, but has been the same since the TMCs were installed about 2 weeks ago. The power is controlled by M80/M81 via a gcode macro for pin ar12 which is inverted to make sure the printer stays off when restarted and the on off values make sense, IE 0=off and 1=on. 

Basically when the printer has been off for a while and then M80 is called the printer will not home until a `RESTART` is issued. Any ideas on this one? It's not a big one, but it's starting to grate on me that every time I switch the printer on I have to issue the restart. 

Log attached:

[klippy.log](https://github.com/KevinOConnor/klipper/files/2227916/klippy.log)

KevinOConnor, 2018-07-25:
What does the ar12 pin (ie, m80) control the power to?  Is this powering off the motors, the stepper drivers, the micro-controller, and/or something else?

When issuing a g28, does the x stepper not move at all, or does it move but it never detects hitting the endstop?  What do you get if you run a DUMP_TMC STEPPER=stepper_x before and after issuing the g28 command?  Does the problem occur on just the x stepper, or does it also occur when running a G28Y0 (or G28Z0)?

-Kevin

AxMod3DPrint, 2018-07-25:
ar12 is the PS_ON pin. That is connected to a SSR which acts as a switch to the PSU which cuts the mains/12v PSU power, the MCU is still powered via USB for the 5v rail, which is always on. 

As for G28 it happens on all steppers, nothing moves. Admittedly this is probably what caused me to check the drivers with Marlin when I first installed them to be sure they were all working correctly. The DUMP_TMC STEPPER=stepper_x(y/z) gives this output, which suggests a clock desync:

`Recv: // GCONF:        0`
`Recv: // GSTAT:        0`
`Recv: // IOIN:        0`
`Recv: // TSTEP:        0`
`Recv: // XDIRECT:        0`
`Recv: // MSCNT:        0`
`Recv: // MSCURACT:        0`
`Recv: // CHOPCONF:        0`
`Recv: // DRV_STATUS:        0`
`Recv: // PWM_SCALE:        0`
`Recv: // LOST_STEPS:        0`

After a power on and a restart the command outputs the values:

`Recv: // GCONF:        4`
`Recv: // GSTAT:        1`
`Recv: // IOIN: 110000f8`
`Recv: // TSTEP:    fffff`
`Recv: // XDIRECT:        0`
`Recv: // MSCNT:        8`
`Recv: // MSCURACT:   f7000c`
`Recv: // CHOPCONF: 14008384`
`Recv: // DRV_STATUS: 80100000`
`Recv: // PWM_SCALE:        0`
`Recv: // LOST_STEPS:        `0`

KevinOConnor, 2018-07-25:
Does it cut the power to the tmc2130 chips though?  (The tmc2130 digital logic can be powered via an internal regulator connected to the motor voltage, or they can be powered via an external 5v rail.)

I'm not sure what you mean by "clock desync"?  There's no indication of a problem with host and micro-controller clock synchronization.  The mcu and tmc2130 don't synchronize their clocks.

-Kevin

CHILLYSMOKES, 2018-07-26:
Dunno if this helps but when I started using TMC2130s I was on Marlin and using the same PS ON/OFF setup with USB powering the mega and driver's wouldn't initiate. Turns out the driver's need power first, before the MCU. So I rigged a powering timer offset so my pi turns on power to the RAMPS first and then the MCU. Haven't had an issue since.

AxMod3DPrint, 2018-07-26:
@KevinOConnor - there is always 5v connected and powered on, this has to be the case else the ps_on will not work. This is provided via the USB connection. I'm assuming here, as I was given the TMCs fully wired by a friend and I may be wrong, that the 5v is powering them internally. As for a de-sync, that was just my interpretation looking at the log, seems I was wrong here. 

@CHILLYSMOKES - does sound like a similar issue, will have to investigate further on that.

AxMod3DPrint, 2018-08-06:
Not as yet, I'm still working around it with power on > restart > home

Not had a chance to look into it further

KevinOConnor, 2018-08-07:
Okay, lets close for now then.

-Kevin

closed 2018-08-07 by KevinOConnor

AxMod3DPrint, 2018-08-19:
Have done a bit of research.. it seems the TMCs require a delay after power is applied via the PS_ON pin, so when M80/SET_PIN is used to pull the pin to the correct value when the default state is off with no delay on sending the settings to the drivers, it's not received, but when a 500ms delay is applied after the power is applied via M80/SET_PIN the drivers are programmed correctly from the config. 

See this issue with Marlin:
https://github.com/MarlinFirmware/Marlin/issues/9058