Klipper3d/klipper #498: !! Error on 'DUMP_TMC': missing STEPPER

StefanMeGit, 2018-07-25:
Hi everyone,

I got some strange problems with my TMC2130 drivers. I use an ArduinoDUE with RADDS and have SPI connected with jumpers to the TMC2130 on my extruder (just for testing!).

My extruder is moving, but I get these error messages from the terminal:

```
Send: DUMP_TMC
Recv: !! Error on 'DUMP_TMC': missing STEPPER

Send: DUMP_TMC STEPPER=extruder
Recv: // GCONF:        0
Recv: // GSTAT:        0
Recv: // IOIN: 11000078
Recv: // TSTEP:    fffff
Recv: // XDIRECT:        0
Recv: // MSCNT:      248
Recv: // MSCURACT:  1200195
Recv: // CHOPCONF: 14028384
Recv: // DRV_STATUS: 81190139
Recv: // PWM_SCALE:        0
Recv: // LOST_STEPS:        0
```
Wiring is definitely ok. I checked three times. Any idea whats wrong? Does the Recv: !! Error on 'DUMP_TMC': missing STEPPER only means I have to add `STEPPER=extruder`?

Thanks!

jakep82, 2018-07-25:
Read the instructions on the g-codes document page.  The first command you sent was missing the name of the driver.  The second command you sent is correct, and it returned the data from the TMC2130 driver that you named extruder.

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

StefanMeGit, 2018-07-28:
Sorry was my mistake... I thought the steppers are not working properly because of an very annoying high pitch noise... after checking the TMC documentation I see that the drivers are fully working! The high pitch noise is another problem.

closed 2018-07-28 by StefanMeGit

stevenvo, 2020-02-16:
> Sorry was my mistake... I thought the steppers are not working properly because of an very annoying high pitch noise... after checking the TMC documentation I see that the drivers are fully working! The high pitch noise is another problem.

what was the prob, mind sharing how you fixed it?