Klipper3d/klipper #499: TMC2130 Problems with Sensorless homing

BKLronin, 2018-07-26:
I can´t get the Sensorless homing to work correctly. On one Axis it worls kind of after loosing 4-5 steps on the end of the axis but on the other with the same motor it just keeps clicking forever  or the sensitivity is so low that it triggers almost imediately. Tried playing with supplied current as well and adjusting SGT value accordingly but no luck there as well. 

So it is too sensitive or not at all barely triggerable with throwing the motor in the opposite direction.

Thanks for any input on this.

[printer.txt](https://github.com/KevinOConnor/klipper/files/2232568/printer.txt)

KevinOConnor, 2018-07-27:
I don't see anything suspicious in the log.  The tmc2130 SGT value is a "magic number" as far as I know - change it (and perhaps other settings) until it (hopefully) works.

You should probably set "homing_retract_dist=0" for the xy axes to disable the second homing bump.  Also, homing at a faster speed (eg, homing_speed=50) may help.

-Kevin

DeltaMaker, 2018-07-28:
Yes, the faster speed improved sensorless homing on our delta 3D printer.  We started with the driver settings from the Prusa config (http://prusaowners.com/wiki/index.php?title=Configuring_Klipper) and adjusted the SGT and run_current a bit.  If you haven't already tried this, it may help.
--Bob

BKLronin, 2018-07-28:
Thank you both very much. 

Tried a lot of combinations of current and SGT values different pins etc. 

Havent tested the speed and found it too slow anyway. Could help, trying..

KevinOConnor, 2018-08-06:
Okay - I'm going to close this for now.  Let us know if you find success.

-Kevin

closed 2018-08-06 by KevinOConnor

jakep82, 2018-08-09:
Voltage also makes a difference.  I found the sensorless homing worked much better when I switched from 12V to 24V.  At 12V they required much higher SGT values, and they were sensitive to minor voltage changes.  A value that worked well with the heated bed off did not work well with the heated bed on because it pulled the voltage down a little.

BKLronin, 2018-08-14:
Changed the homing speed to 30mm/s at a 3 SGT value. Works flawlessly every time. 

Printing looks good with the firmware so far :)

I went with 12 V at the beginning but quickly changed to 24V after reading more about the drivers. They don´t like 12 V at all it seems .