Klipper3d/klipper #501: Z - PROBE on wrong MCU

sniperlucian, 2018-07-26:
I have Mulit MCU and a 3 stepper Z Achsis.

First MCU:  
COREXY,  Xstepper, Ystepper and Estepper

SECOND MCU:
Stepper Z1, Stepper Z2, Stepper Z2  for Z Achsis

I am trying to get the Z_TILT_ADJUST feature running. However, when I define the [PROBE] to be on the SECOND MCU,  klipper complains that the PROBE needs to be on the same MCU as the stepper.

**### CONFIG:ERROR -  Endstop and stepper must be on the same mcu**

Its my understanding that the Z-Probe is for the Z-AXIS stepper - and therefore the SECOND CPU would be the right one !?!?

[printer.txt](https://github.com/KevinOConnor/klipper/files/2232897/printer.txt)

KevinOConnor, 2018-07-27:
If you're encountering an issue, you must attach the klipper log file.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

sniperlucian, 2018-07-27:
Ok - here it is. ;)

[klippy.log](https://github.com/KevinOConnor/klipper/files/2236498/klippy.log)

KevinOConnor, 2018-07-28:
Change your endstop pin configuration to:
`endstop_pin: probe:z_virtual_endstop`
as described in config/example-extras.cfg.

-Kevin