Klipper3d/klipper #502: Timeout on serial connect

SergeyKrutyko, 2018-07-27:
SW: OrangePi
HW: Atmega2560 + RAMPS 1.4
made all as manuals. 
sucessfuly flash atmega
Klipper installed as user dsi
user included in tty and dialout groups.
when i start klipper see next:

===== log ========
Starting Klippy...
Args: ['/home/dsi/klipper/klippy/klippy.py', '/home/dsi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.6.0-307-g7c7de85'
CPU: 4 core ARMv7 Processor rev 5 (v7l)
Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
Start printer at Fri Jul 27 04:25:12 2018 (1532654712.9 1425.2)
===== Config file =====
[stepper_x]
step_pin = ar54
dir_pin = ar55
enable_pin = !ar38
step_distance = .0125
endstop_pin = ^ar3
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_y]
step_pin = ar60
dir_pin = !ar61
enable_pin = !ar56
step_distance = .0125
endstop_pin = ^ar14
position_endstop = 0
position_max = 190
homing_speed = 50

[probe]
pin = ^!ar18
z_offset = -0.62
speed = 5.0

[stepper_z]
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = .00083
endstop_pin = probe:z_virtual_endstop
position_endstop = 0
position_max = 120

[bed_tilt]
x_adjust = 0
y_adjust = -0
points = 25,25
        25,95
        25,165
        100,165
        100,95
        100,25
        175,25
        175,95
        175,165

[extruder]
step_pin = ar26
dir_pin = ar28
enable_pin = !ar24
step_distance = .0023754
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = ar10
sensor_type = NTC 100K beta 3950
pullup_resistor = 2000
sensor_pin = analog13
control = pid
pid_kp = 19.72
pid_ki = 1.9
pid_kd = 51.29
min_temp = 0
max_temp = 270

[heater_bed]
heater_pin = ar8
sensor_type = NTC 100K beta 3950
pullup_resistor = 2000
sensor_pin = analog14
control = watermark
min_temp = 0
max_temp = 130

[fan]
pin = ar9

[display]
lcd_type = st7920
cs_pin = ar16
sclk_pin = ar23
sid_pin = ar17

[mcu]
serial = /dev/ttyUSB0
pin_map = arduino
baud = 250000

[mcu host]
serial = /tmp/printer

[printer]
kinematics = corexy
max_velocity = 300
max_accel = 3000
max_z_velocity = 25
max_z_accel = 30

=======================
Extruder max_extrude_ratio=0.266081
Starting serial connect
Sending MCU 'mcu' printer configuration...
Configured MCU 'mcu' (634 moves)
Loaded MCU 'mcu' 69 commands (v0.6.0-307-g7c7de85-20180727_042431-octoprint / gcc: (GCC) 4.9.2 binutils: (GNU Binutils) 2.26.20160125)
MCU 'mcu' config: RECEIVE_WINDOW=192 SOFT_PWM_MAX=256 SERIAL_BAUD=250000 ADC_MAX=1023 PWM_MAX=255 CLOCK_FREQ=16000000 MCU=atmega2560 STATS_SUMSQ_BASE=256
Starting serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

=== end log ===

return firmware of marlin - all worked perfectly :(
i dont know how need made for start klipper :(

AxMod3DPrint, 2018-07-27:
Check this from the FAQ documents - https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

SergeyKrutyko, 2018-07-27:
offcoz i try it - no effects.

avrdude flash my mcu via /dev/ttyUSB0 without problems.... octoprint working with marlin wo problems too............

AxMod3DPrint, 2018-07-27:
If it's not connecting then it's a port issue. I would ask that you try the serial by ID method again, re-flashing with the serial ID and not /tty/USB0. This should _always_ connect no matter what happens.

SergeyKrutyko, 2018-07-27:
i try it now by id - problem not solved............ flashing sucessfully but klipper still connect timeout :(

AxMod3DPrint, 2018-07-27:
That suggests klipper has stopped or octoprint can't see the port for some reason.

Check klipper is running - `sudo service klipper status` 

Ok, check /tmp/printer exists on the OrangePi - `cd /tmp/ ; ls -al | grep printer`

I would recommend a full restart of klipper just to make sure - `sudo service klipper restart`

If octoprint still doesn't connect after that, restart octoprint, this can be done via the GUI, if you have it set up or via `sudo service octoprint restart` 

That should generally resolve any connectivity issues with Klipper and Octoprint.

SergeyKrutyko, 2018-07-27:
you dont right.....  octoprint could not connecto to klipper, while klipper not connect to printer. klipper reset connection from octaprint, because mcu not connected!

SergeyKrutyko, 2018-07-27:
sudo service klipper start
dsi@octoprint:~/klipper$
dsi@octoprint:/tmp$ sudo service klipper status
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated; vendor preset: enabled)
   Active: active (running) since Fri 2018-07-27 13:44:07 MSK; 3min 33s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 12744 ExecStop=/etc/init.d/klipper stop (code=exited, status=0/SUCCESS)
  Process: 12831 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCESS)
    Tasks: 6 (limit: 4915)
   CGroup: /system.slice/klipper.service
           └─12836 /home/dsi/klippy-env/bin/python /home/dsi/klipper/klippy/klippy.py /home/dsi/printer.cfg -l /tmp/klippy.log

Jul 27 13:44:06 octoprint systemd[1]: Starting LSB: Klipper daemon...
Jul 27 13:44:07 octoprint klipper[12831]: Starting klipper: klipper.
Jul 27 13:44:07 octoprint systemd[1]: Started LSB: Klipper daemon.

SergeyKrutyko, 2018-07-27:
cd /tmp/ ; ls -al | grep printer
lrwxrwxrwx  1 dsi  dsi     10 Jul 27 13:44 printer -> /dev/pts/4

SergeyKrutyko, 2018-07-27:
in logs klippy.log i see "printer is not ready" string when octoprint try connect

SergeyKrutyko, 2018-07-27:
probles still between klipper and mcu connection!

SergeyKrutyko, 2018-07-27:
now i try flash mcu by marlin with bootloader, via octoprint. marlin stated and all worked. i try again flash klipper mcu flash - and again connection timeoud from klipper to mcu

SergeyKrutyko, 2018-07-27:
already made it at install:

dsi@octoprint:/tmp$ getent group tty
tty:x:5:dsi
dsi@octoprint:/tmp$ getent group dialout
dialout:x:20:dsi
dsi@octoprint:/tmp$

AxMod3DPrint, 2018-07-27:
Ok, about the only thing I can suggest at the moment is to _attach_ the log to this thread as per contact.md

SergeyKrutyko, 2018-07-27:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2235625/klippy.log)

[printer.zip](https://github.com/KevinOConnor/klipper/files/2235626/printer.zip)
[config.zip](https://github.com/KevinOConnor/klipper/files/2235628/config.zip)

KevinOConnor, 2018-07-27:
Remove the "[mcu host]" section from your config.  The "mcu" sections describe micro-controllers (see config/example-extras.cfg).  Don't try to configure the output of Klipper (/tmp/printer) in the config file.

-Kevin

SergeyKrutyko, 2018-07-29:
thanks! all started!

closed 2018-08-06 by KevinOConnor