Klipper3d/klipper #504: Option 'homing_endstop_align_zero' is not valid in section 'stepper_z'

StefanMeGit, 2018-07-28:
I tried to setup the homing_endstop_align_zero function, but get an error message... 

Did I misunderstood something? 

```
# Radds 1.6
# Temp sensor pins: analog0, analog1, analog2, analog3, analog4 (Heatbed)
# Mosfet Pins: ar7 (Heatbed), ar8, ar9, ar11, ar12, ar13

[stepper_x]
step_pin: ar24
dir_pin: ar23
enable_pin: !ar26
step_distance: .012363
endstop_pin: ^ar34
position_endstop: 290
position_max: 290
homing_positive_dir: true
homing_speed: 50
homing_retract_dist: 2.0

[stepper_y]
step_pin: ar17
dir_pin: ar16
enable_pin: !ar22
step_distance: .012363
endstop_pin: ^ar36
position_endstop: 210
position_max: 210
homing_positive_dir: true
homing_speed: 50
homing_retract_dist: 2.0

[stepper_z]
step_pin: ar2
dir_pin: ar3
enable_pin: !ar15
step_distance: .00126
endstop_pin: ^ar32
position_endstop: 0
position_max: 210
homing_positive_dir: false
homing_speed: 3
homing_retract_dist: 1.0
homing_stepper_phases: 64
homing_endstop_align_zero: true
# Stepper phase adjusted endstops. The following additional parameters
# may be added to a stepper axis definition to improve the accuracy of
# endstop switches.
#[stepper_z]
#homing_stepper_phases:
#   One may set this to the number of phases of the stepper motor
#   driver (which is the number of micro-steps multiplied by
#   four). This parameter must be provided if using stepper phase
#   adjustments.
#homing_endstop_accuracy: 0.200
#   Sets the expected accuracy (in mm) of the endstop. This represents
#   the maximum error distance the endstop may trigger (eg, if an
#   endstop may occasionally trigger 100um early or up to 100um late
#   then set this to 0.200 for 200um). The default is
#   homing_stepper_phases*step_distance.
#homing_endstop_phase:
#   This specifies the phase of the stepper motor driver to expect
#   when hitting the endstop. Only set this value if one is sure the
#   stepper motor driver is reset every time the mcu is reset. If this
#   is not set, then the stepper phase will be detected on the first
#   home and that phase will be used on all subsequent homes.
#homing_endstop_align_zero: False
#   If true then the code will arrange for the zero position on the
#   axis to occur at a full step on the stepper motor. (If used on the
#   Z axis and the print layer height is a multiple of a full step
#   distance then every layer will occur on a full step.) The default
#   is False.

[stepper_z1]
step_pin: analog10
dir_pin: !analog9
enable_pin: !analog11
step_distance: .00126

[extruder]
step_pin: analog7
dir_pin: analog6
enable_pin: !analog8
step_distance: .01061
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar12
min_extrude_temp: 20
max_extrude_only_distance: 601
max_extrude_cross_section: 50
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog0
control: pid
pid_Kp: 23.715
pid_Ki: 1.189
pid_Kd: 118.281
min_temp: 15
max_temp: 270

[tmc2130 extruder]
cs_pin: ar25
microsteps: 16
interpolate: true 
run_current: 0.65
hold_current: 0.65
sense_resistor: 0.110
stealthchop_threshold: 0

[heater_bed]
heater_pin: ar7
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog4
control: pid
min_temp: 15
max_temp: 130
pid_Kp: 64.498
pid_Ki: 0.988
pid_Kd: 1052.126

[fan]
pin: ar9

[heater_fan nozzle_cooling_fan]
pin: ar8
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

[mcu]
serial: /dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_55730323831351418181-if00
baud: 250000
pin_map: arduino

#[mcu secondRADDS]
#serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
#baud: 250000
#pin_map: arduino

[printer]
kinematics: corexy
max_velocity: 300
max_accel: 3000
max_z_velocity: 50
max_z_accel: 100

[homing_override]
gcode:
 G90
 G1 Z1 F100
 G28 Y
 G28 X 
 G28 Z
#set_position_x: 100
#set_position_y: 100
set_position_z: 0
```

KevinOConnor, 2018-07-30:
Currently, that option is only available if the homing_endstop_phase is also specified.  (It's that way due to the implementation - the code internally adjusts the configured endstop_position with the given endstop phase so that Z0 is at a full step.)

In the future, please attach the klipper log to any issues - the config doesn't help nearly as much.

-Kevin

StefanMeGit, 2018-07-31:
Everything works now as expected! Great option! 

Just a question for stepper_z1. Does Klipper handle the second stepper also to home on a full steps if I activate this on stepper_z?

KevinOConnor, 2018-08-01:
If the two stepper drivers are reset at the same time, if both stepper drivers have the same step_distance, both stepper drivers are using the same endstop, and z_tilt is not used - then both stepper drivers should always be in the same phase.  So, yes, in that situation they would both be at a full step at z=0.  (In any other situation, the steppers would likely not be in phase.)

-Kevin

StefanMeGit, 2018-08-01:
Thanks a lot Kevin!

closed 2018-08-01 by StefanMeGit