Klipper3d/klipper #505: T1, that tool doesn't exist according to the printer profile

bruce356, 2018-07-29:
Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6b88d970, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.6.0-361-gaebc1f5 FIRMWARE_NAME:Klipper
Send: N2 M21*18
Recv: ok
[...]
Send: status
Recv: // Printer is ready
Recv: ok
[...]
Not queuing T1, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware
Send: T1
Recv: ok
Not sending T1, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware
[...]
Send: T0
Recv: ok
[...]

Send
[klippy.log](https://github.com/KevinOConnor/klipper/files/2238491/klippy.log)

mamoit, 2018-07-30:
Does your printer profile in octoprint have 2 extruders?
You need to change it too there.

KevinOConnor, 2018-07-30:
There doesn't appear to be anything unusual in the logs.  The messages quoted above appear to be a warnings from OctoPrint - fix by configuring OctoPrint accordingly (as suggested by @mamoit ).

-Kevin

Tahutipai, 2020-11-26:
Just for completeness: this will also occur if you have updated Octoprint, but not restarted it.
(Octoprint doesn't advise a restart is required when adding an additional extruder, but a restart is required)