Klipper3d/klipper #506: Dual Nozzle mode

mamoit, 2018-07-31:
Marlin has [this neat feature](http://marlinfw.org/docs/gcode/M605.html), where T1 can either mirror or duplicate the moves of T0.
It would be quite an advanced option to have, but a very nifty one for dual extruder printers.
Ideas?

KevinOConnor, 2018-08-06:
I'm sure it could be implemented in Klipper.  It would require software development work.  If you are interested in tackling that, then I recommend you introduce a new "extras" module to add that support - see the docs/Code_Overview.md file for the details.

-Kevin

mamoit, 2018-08-07:
I don't know if I'll be able to do this (at least in the next couple of weeks).
I'm still getting used to the codebase and have some other high priority stuff to do.

In the meantime, if somebody has ideas/suggestions, let us know.
Thanks :+1:

KevinOConnor, 2018-08-16:
Okay.  I'll close this for now.  I can reopen if you start development and have questions.

closed 2018-08-16 by KevinOConnor

mamoit, 2018-08-17:
You could label it as enhancement and leave it open :slightly_smiling_face: 
Or you prefer to have a clean issue list?