Klipper3d/klipper #508: Z-Axis Homing causes "MCU 'mcu' shutdown: Timer too close"

pappy15, 2018-07-31:
Hello,
I started with a clean octoprint and klipper install (on raspberry pi 2) and everything seems to work fine. Only the Z-Axis doesn't work. If I want to home the z-axis, I get the error : 

> MCU 'mcu' shutdown: Timer too close This is generally indicative of an intermittent communication failure between micro-controller and host. Once the underlying issue is corrected, use the "FIRMWARE_RESTART" command to reset the firmware, reload the config, and restart the host software. Printer is shutdown

I have an ultimaker clone with two motors on z-axis which use one stepper driver (arduino mega 2560 + ramps 1.4 board + drv8825 on 1/32 step).
On marlin it worked with this setup, but I want to test klipper now.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2245543/klippy.log)

pappy15, 2018-07-31:
I think I found a solution: I tested with 1/16 microstepping on z-axis driver and now I can home all axis.

KevinOConnor, 2018-08-01:
It looks like the attached log is a snippet from the log.  We need the full log to be attached in order to help.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

closed 2018-08-09 by KevinOConnor