Klipper3d/klipper #509: Timeout on serial connect on Ender 3

chrnueve, 2018-08-01:
Hello,
I'm facing an error trying to make this run on a Raspberry Pi 3 with Octoprint and a Creality Ender 3 (atmega1284p based).
I've burned the bootloader and flashed the firmware with the default settings (attached as
[config.txt](https://github.com/KevinOConnor/klipper/files/2247481/config.txt) )
After creating the /tmp/printer link on Octoprint, copying the printer.cfg from config/printer-creality-ender3-2018.cfg and starting the klipper service I see this on /tmp/klippy.log:

    Starting Klippy...
    Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
    Git version: 'v0.6.0-308-gac5e207'
    CPU: 4 core ARMv7 Processor rev 4 (v7l)
    Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
    Start printer at Tue Jul 31 23:52:23 2018 (1533081143.1 511.5)
    (//prints config file)
    =======================
    Extruder max_extrude_ratio=0.266081
    Starting serial connect
    Timeout on serial connect
    Timeout on serial connect
    Timeout on serial connect
    (...)

Stoping klipper service and using the console.py with `./klippy-env/bin/python ./klipper/klippy/console.py /dev/ttyUSB0 115200`:

    ==================== attempting to connect ====================
    INFO:root:Starting serial connect
    DEBUG:root:Starting stk500v2 leave programmer sequence
    DEBUG:root:Got '' from stk500v2
    Loaded 69 commands (v0.6.0-308-gac5e207-20180731_230700-octopi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
    MCU config: RECEIVE_WINDOW=192 SOFT_PWM_MAX=256 SERIAL_BAUD=115200 ADC_MAX=1023 
    PWM_MAX=255 CLOCK_FREQ=16000000 MCU=atmega1284p STATS_SUMSQ_BASE=256
    DEBUG:root:new minimum rtt 1091.957: hrtt=0.000582 freq=16000000
    ====================       connected       ====================
    003.350: stats count=122 sum=220547 sumsq=2493798
    008.446: stats count=56 sum=28904 sumsq=126794
    013.543: stats count=56 sum=28630 sumsq=125781
    018.545: stats count=55 sum=28325 sumsq=122534
    (...)

Sending a STATS command:
    STATS
    bytes_write=1193 bytes_read=4992 bytes_retransmit=9 bytes_invalid=0 send_seq=177 receive_seq=177 retransmit_seq=2 srtt=0.010 rttvar=0.004 rto=0.027 ready_bytes=0 stalled_bytes=0 freq=16000295
    109.353: stats count=56 sum=28621 sumsq=124924
    DEBUG:root:new minimum rtt 1199.766: hrtt=0.000218 freq=16000294
    114.450: stats count=57 sum=30796 sumsq=160391

The message in `DEBUG:root:Got '' from stk500v2 `changes on every run, for example `DEBUG:root:Got '\x05\x13\xac\x1a~' from stk500v2`

Everyting on RPi is as it comes from Octopi fresh install, and reflashed the firmware twice

unnefer1, 2018-08-01:
Hi. 

Have you tried setting the baudrate used during menu config in your printer.cfg? Also, are you using the device name, try using the device id instead (also in printer.cfg). 

To find the device id, type the following into the rpi's console:
ls -l /dev/serial/by-id/

This will print out a line similar to this:
lrwxrwxrwx 1 root root 13 Jun  1 21:12 usb-1a86_USB2.0-Serial-if00-port0 -> ../../ttyUSB0

You then add the baudrate you set during menu config and the the device ID to your printer.cfg file under the MCU section:
[mcu]
baud: 250000
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

Note: If the above doesn't work, this might sound silly, but try setting the baudrate to 250000 during menu config and then use the single make/flash command with the device id to flash Klipper:
make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

You'll then need to update printer.cfg and change baudrate to match:
[mcu]
baud: 250000
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

AxMod3DPrint, 2018-08-01:
Please provide your log file as described in https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

chrnueve, 2018-08-01:
I've changed the baudrate and now connects correctly!.
Now I'm facing a problem with pin names and BLTouch, should I close this Issue and open another one?
Here's the current klippy.log, sorry @AxMod3DPrint, the klippy.log for the previous iteration was inline 
[klippy.log](https://github.com/KevinOConnor/klipper/files/2248700/klippy.log)

Thank you!

unnefer1, 2018-08-01:
Hi chrnueve. I'm glad it is now connecting for you :)

I think it would be best to close this one off and log the other problem you're having separately with a heading/subject that suits that additional problem.

chrnueve, 2018-08-01:
Thank you @unnefer1, closing this issue, I'm going to try that 👍

closed 2018-08-01 by chrnueve