Klipper3d/klipper #510: Error trying to implement  "force_move"

bruce356, 2018-08-01:
Send: status
Recv: // Error loading kinematics 'cartesian'
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: !! Error loading kinematics 'cartesian'
Recv: ok
[...]

[klippy.log](https://github.com/KevinOConnor/klipper/files/2248071/klippy.log)

Thanks

AxMod3DPrint, 2018-08-01:
I believe force_move is a newer addition, you are using @mcmatrix branch for the LCD menu, it is quite possible that that isn't currently in line with the master branch in terms of latest additions and hasn't been rebased to the latest commits. As such, force_move may not be implemented, and going by the log, this seems to be possibly the case, although I could be wrong. 

I'll leave @mcmatrix or @KevinOConnor to weigh in here.

KevinOConnor, 2018-08-01:
Right - the code being run isn't from a branch in this repo.  This issue tracker is only for support of code from this repo.

mcmatrix, 2018-08-01:
Sorry for this, but that repo is just for menu testing purposes (not meant for normal usage).
Currently, I'm working hard to merge the menu module to the Klipper master branch.
@bruce356 You have to wait until the merge is done. Then the menu module will
be part of Klipper.

KevinOConnor, 2018-08-01:
To be clear, the code being run doesn't appear to be from the menu testing branch either.

closed 2018-08-16 by KevinOConnor