Klipper3d/klipper #511: multiple control boards (query, not bug)

unnefer1, 2018-08-01:
Hi,

When going through the code and the documentation, I saw that Klipper supports multiple control boards, which is a great thing.

What I wanted to know, is if the control boards all need to be the same, or if I can use any mix of control boards together?

ie. I have a Tronxy x5s that uses a Chitu 'Melzi' board that is an atmega1284p. I have an e3d Kraken on the way, but the board has no spare drivers or expansion pins available.

Can I simply add a MKS v1.4 (uses an atmega2560r3) to run the 3 additional extruders in unison with the Chitu board I currently have?

Or do both boards need to run the same chip, which would thenmean removing the stock Chitu board and running 2x MKS v1.4 boards in its place?

Feedback appreciated.
nef

CHILLYSMOKES, 2018-08-01:
Any mix that is supported. I use a 32 bit Re-Arm and an 8 bit MEGA. Just make sure you are "make"ing each properly with correct serial call out during MCU flash and addressing each component on the non-primary MCU in your config. Look at the example config for multiple MCU for syntax. Good luck.

StefanMeGit, 2018-08-01:
I also used an Arduino DUE and an MKS GEN 1.4 together. No issues with that! Feel free to combine whatever MCU is supported.

There is only an issue with endstops as I know. They have to be on the main MCU as well as the the homing axis... but maybe this also changed.

unnefer1, 2018-08-01:
Thanks for the replies. Greatly appreciated.

Re: main mcu. Should the controller with the faster chip be the primary, or does that not matter since the rpi takes care of the hard processing?

Since the Chitu controller in the Tronxy x5s is so limited on additional features (it's a very basic, bare minimum contro board), does that mean I'll need to set the MKS v1.4 control board as the main mcu if I want to take advantage of the additional min/max end stops ait includes?

StefanMeGit, 2018-08-01:
I have to revoke my answer! 
U can use all axis freely on every board u want, but the endstop for the specific axis should be on the same board as the driver.

In my understanding it doesn’t matter what mcu is the “main”-mcu.

But maybe someone can give a more knowledgeable answer on this?

KevinOConnor, 2018-08-01:
It's recommended that the main mcu be the mcu with the steppers controlling the X and Y axes.  It is recommended that the X and Y axes be on the same MCU, but if for some reason one doesn't want to do that, it would be recommended that the MCUs controlling the X and Y be of the same type of MCU.  It is also recommended that if there are multiple stepper drivers controlling a single axis (eg, dual z steppers) then all of them should be on the same mcu.  The above recommendations should improve timing (at the sub-millisecond level).

As others have stated, there are few "hard limits" on multiple mcu use.  The one restriction that comes up periodically is the requirement that the endstops associated with a stepper must be on the same mcu as that stepper.

To answer the original question - it should be fine to add additional extruders to an additional mcu board with an atmega2560.  Leave the main mcu as the melzi (that controls the XY).

-Kevin

unnefer1, 2018-08-04:
Thanks all for the info. Will close this one now.

closed 2018-08-04 by unnefer1