Klipper3d/klipper #514: unable to set negative position_endstop on Z-axis

salfter, 2018-08-02:
I'm trying to get Klipper running on an Anet A8 with a RAMPS+Arduino Mega combo and a 3DTouch.  I need to set a negative Z-probe offset because the probe's zero is about 2.2 mm below the nozzle, but Klipper bombs out immediately:

```
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.6.0-310-gf2d232e'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
Start printer at Thu Aug  2 09:42:18 2018 (1533228138.6 525890.9)
===== Config file =====
[stepper_x]
step_pin = ar54
dir_pin = ar55
enable_pin = !ar38
step_distance = .005
endstop_pin = ^!ar3
position_endstop = 0
position_max = 220
homing_speed = 50

[stepper_y]
step_pin = ar60
dir_pin = ar61
enable_pin = !ar56
step_distance = .0125
endstop_pin = ^!ar14
position_endstop = 0
position_max = 220
homing_speed = 50

[stepper_z]
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = .00125
endstop_pin = ^!ar18
position_endstop = -2.2
position_max = 240

[extruder]
step_pin = ar26
dir_pin = ar28
enable_pin = !ar24
step_distance = .005
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = ar10
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog13
control = pid
pid_kp = 28.35
pid_ki = 2.37
pid_kd = 84.76
min_temp = 0
max_temp = 250

[heater_bed]
heater_pin = ar8
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog14
control = pid
pid_kp = 275.85
pid_ki = 33.67
pid_kd = 564.93
min_temp = 0
max_temp = 130

[fan]
pin = ar9

[mcu]
serial = /dev/ttyACM0
pin_map = arduino

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100

[display]
lcd_type = st7920
cs_pin = ar16
sclk_pin = ar23
sid_pin = ar17

[servo bltouch]
pin = ar7
maximum_servo_angle = 180
minimum_pulse_width = 0.0006
maximum_pulse_width = 0.0024

[probe]
pin = !ar19
z_offset = 2.345
activate_gcode = 
	SET_SERVO SERVO=bltouch ANGLE=10
	SET_SERVO SERVO=bltouch ANGLE=60
	G4 P200
deactivate_gcode = 
	SET_SERVO SERVO=bltouch ANGLE=90
	G4 P100

=======================
Error loading kinematics 'cartesian'
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/toolhead.py", line 245, in __init__
    self.kin = mod.load_kinematics(self, config)
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 165, in load_kinematics
    return CartKinematics(toolhead, config)
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 14, in __init__
    for n in ['x', 'y', 'z']]
  File "/home/pi/klipper/klippy/stepper.py", line 265, in LookupMultiRail
    rail = PrinterRail(config)
  File "/home/pi/klipper/klippy/stepper.py", line 143, in __init__
    " position_min and position_max" % config.get_name())
Error: position_endstop in section 'stepper_z' must be between position_min and position_max
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 242, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 220, in _read_config
    m.add_printer_objects(config)
  File "/home/pi/klipper/klippy/toolhead.py", line 452, in add_printer_objects
    config.get_printer().add_object('toolhead', ToolHead(config))
  File "/home/pi/klipper/klippy/toolhead.py", line 249, in __init__
    raise config.error(msg)
Error: Error loading kinematics 'cartesian'
```

KevinOConnor, 2018-08-02:
If you need to set a negative position_endstop, then you must also explicitly set a negative position_min.  So, add the following to your stepper_z section:
```
position_min: -2.2
```

-Kevin

salfter, 2018-08-02:
I found something similar on other axes in the bundled A8 config (which it also turned out I needed to set to get them running right).

closed 2018-08-02 by salfter