Klipper3d/klipper #516: bowden hotend + pressure advance  = does nothing for oozing or blobs

unnefer1, 2018-08-04:
Quick hardware summary:
Tronxy x5s (corexy) with ~900mm bowden tube
e3d v6 all-metal (bowden stops before heatbreak)
0.4mm nozzle
Cura "draft quality" profile (0.2 layer height, 0.4 line widths)

Problem: 
Have ooze blobs at the start of each new line and start of each new layer and pressure_advance setting doesn't seem to do anything for my hardware setup - iare bowden printers not compatible with pressure advance?

I've tried about 50 different combinations of retraction from 0 to 20mm and coasting from off to default 0.064mm3 to a huge 2mm3. Nothing has any impact.

From watching as it happens, it looks like the hotend stops momentarily at every corner and at the start and end of every line, which allows a little filament to ooze out, causing the blobs.

Any ideas on what setting in printer.cfg I need to adjust?

Below is a photo of "square fish" using pressure_advance values from 0.05 through to 0.8 with no change to the blobs:
[image]

kevinmce, 2018-08-04:
Check that your extruder max velocity is high enough. Also check your max acceleration (can only be set for all xyze combined) 
If either of these are too low the printer will end up spending too much time on line start / stops. 
An additional thing to check is your extruder esteps, make sure its not overextruding.

AxMod3DPrint, 2018-08-04:
0.8 seems to be getting better. Bowden systems can go up to 2.0, so you can push it further, above 2.0 you won't see any improvement. As a note, make sure settings such as coast and wipe are off in your slicer for this to work. 

All our printers apart from 1 are Bowden, and all run the same filament extruder drive train (Drakon Extruder into an E3D v6) and we found there wasn't much improvement, so we went back to Coast and Wipe, which effectively does the same thing, but in the slicer. In our tests the Direct drive benefited more from pressure advance than the Bowden. So YMMV with it.

dragonnn, 2018-08-04:
Not about klipper but they is a mechanical solution to reduce the accumulation of pressure in the bowden tube. Check this out https://www.facebook.com/groups/Hypercube.Evo/permalink/192106034761003/
For me it made a really huge difference.

BlackStump, 2018-08-04:
I found for extruder only moves ie retracts I needed to increase these settings to minimise the pause on retracts.
ie
#max_extrude_only_velocity:
#   Maximum velocity (in mm/s) of the extruder motor for extrude only
#   moves. If this is not specified then it is calculated to match the
#   limit an XY printing move with a max_extrude_cross_section
#   extrusion would have.
#max_extrude_only_accel:
#   Maximum acceleration (in mm/s^2) of the extruder motor for extrude
#   only moves. If this is not specified then it is calculated to
#   match the limit an XY printing move with a
#   max_extrude_cross_section extrusion would have.

my settings at the moment
max_extrude_only_velocity: 120
max_extrude_only_accel: 1500

BlackStump, 2018-08-04:
yikes copy and paste is a tad loud

unnefer1, 2018-08-04:
Hahaha no worries BlackStump. Thank you for that information. I will try them when I get a moment spare today.

unnefer1, 2018-08-04:
P.S. Do I add these under the [printer] section or the [extruder] section in my printer.cfg?

CHILLYSMOKES, 2018-08-05:
@BlackStump do you use a Bowden or direct drive extruder?? Thanks.

BlackStump, 2018-08-05:
@unnefer1  under [extruder]

@CHILLYSMOKES    I have a Bowden

cheers

BlackStump, 2018-08-05:
@KevinOConnor  just curious would a firmware retract implementation make more sense with pressure advance?
if so then a on the fly adjustment for retract would make fine tuning much easier.

BlackStump, 2018-08-05:
Pressure advance in action at the extruder

https://youtu.be/5-gDjmpJQ_g

cmock, 2018-08-05:
My 2¢: 

On my Tronxy X5S I've given up on pressure advance; It starts to improve quality at 0.2, but the extruder (Titan clone) also starts to lose steps at that setting. I've since turned it off, and the next thing I'll probably try is capricorn bowden tube.

On my Kossel Mini, OTOH, with a direct drive extruder and 570mm of bowden length, I'm pretty happy at pressure_advance=0.15.

unnefer1, 2018-08-06:
I've had to revert back to Repetierfor now as I have a few print jobs I need to complete by the end of the week.  After they are done, I'll flash  Klipper again and try again.

FYI: I use separate sdcards with Repetier server and Octoprint on them so I can quickly swap back and forth. The odd thing is that when I swap out the Octoprint sdcard for the Repetier server sdcard, those pauses that cause the blobs are no longer present, so the blobs are kept to a minimum.

Anyway, will post again on the weekend.

AxMod3DPrint, 2018-08-06:
Just as a note, Repetier server supports Klipper, so you shouldn't really need to switch over to Octoprint if you don't need to. 

Also, the only reason I found for having those blobs whilst Pressure Advance was enabled was due to having Coast/Wipe/Z Hop enabled in the slicer, as soon as those were disabled the issue disappeared. I'd double check your slicer settings again.

KevinOConnor, 2018-08-06:
I can only provide some high level thoughts - every printer is different so it is hard to say what the issue is in the original post.

I fear the "square fish test" is not a useful method of testing pressure advance - the blobs seen in that test may be due to initial nozzle/bed z distance inaccuracies, z step distance issues, e step distance issues, retraction speed, slicer move planning, warping, etc.  If you're looking to tune pressure advance, I recommend following the guide at: https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md and then posting pictures from that test.

Since writing the pressure advance tuning document, I have found it is possible to go to much higher pa values if one also decreases acceleration.  For example, my delta (with an ~1m long bowden) works well with pa=0.500.  I have yet to have time to update that document.

I suspect coasting will result in poor quality on klipper - disable it.  Wipe on retract is useful.  I do find retract to also be useful in combination with pa, but ony with small retraction values (eg, 0.5 mm).

If one is using very large retraction values, it may be necessary to tune max_extrude_only_velocity/accel - the defaults result in conservative values (that I've found improve quality when small retractions are used), but they wont enable sudden massive filament shifts that some large retract settings require.

-Kevin

unnefer1, 2018-08-06:
Thanks fopr the reply @KevinOConnor 

For some reason the square.stl file would not load for me after I downloaded it, so I used the "square fish test" stl as it was suggested to tune linear advance in Marlin.

I totally agree it could be a number of things causing the blobs, but switching to a different firmware removes the constant pauses that cause the blobs, so I can only conclude it has to have something to do with the Klipper & Octoprint combination.

I actually tried every setting in Cura & Slic3r that I could think of over a full day, including:

- retraction on/off (including trying different retraction subsettings)
- coasting on/off
- wipe on/off
- hotend temps from 190c-215c
- bed temps from 40c-60c
- z-offset from 0-0.3
- z-hop on/off
- print speed from 50mm/s-120mm/s
- cooling fan from of to 100%

I also tried the following in Klipper printer.cfg:

- pressure_advance from 0.05 to 0.8 (as per photos in first post
- max_velocity from 200-400
- max_accel from 1000-3000
- max_z_velocity from 5-20
- max_z_accel from 20-100

I even tried diffent hardware things:

- belts tight
- belts loose
- shortened bowden tube (buy 20cm)
- adjust filament extruder tension
- reset z-height and relevelled bed
- adjusted parts cooling fan to be point further below the nozzle tip

As you can see, I've tried everything, and no matter what I tried, pressure advance just didn't seem to have any impact on the final print.

Current conclusion from the above (in no particular order):
- Pressure Advance is simply not running
- Pressure Advance is running, but is not working with the bowden tube length used by Tronxy x5s
- Tronxy x5s filament extruder is faulty
- hotend heatsink fan (not parts cooling fan) is not cooling heatsink enough and there is heat soak that is causing oozing
- nozzle is not the size that it states
- A problem with serial communication on my rpi, causing it to constantly pause (maybe a plugin)
- A problem with Klipper & Octoprint combination on my rpi

I'll give it another shot this coming weekend and see if I can rule out some of the above conclusions.

oderwat, 2018-08-06:
You can easily see if PA is running. The extruder will have extra movement for and back while printing something like a large rectfill area.

I think the "pauses" you mention may be the culprit for your observations. 

Still I wrote an issue some weeks ago which basically tests PA for my case and has similar complains as yours. See: https://github.com/KevinOConnor/klipper/issues/361 ... 

You may want to try the GCode I posted there (check if it can be printed on your printer first). It will probably not adhere on your printing plate. I can easily print single walls on mine. Still, the GCode is part of this comment: https://github.com/KevinOConnor/klipper/issues/361#issuecomment-394530532 .. It automatically changes PA at two key positions. I could get my printer to not make visible seems with that. It also prints the "rectangles" very nice. Real prints still have seems everywhere though. Kinda like what you described.

unnefer1, 2018-08-20:
I wanted to post an update as I had an opportunity to better test and calibrate pressure advance for the Tronxy x5s over the past weekend.

After spending the better part of both days of the weekend trying to stop the blobs from occuring at the start of each layer and/or line segment by using pressure advance, I've come to the following conclusions:
- pressure advance alone does not fix the blobbing 
- retraction alone does not fix the blobbing
- the "max_extrude_only_velocity" and "max_extrude_only_accel" settings suggested by BlackStump also did not fix the blobbing
- combinations of the above does not fix the blobbing

The only way to reduce the blobbing at the start of each layer and/or line segment, which I am still not exactly sure is the best solution to fix the problem, was to first:
- dramatically increase extruder acceleration
- greatly increase travel speed in the slicer software
- use very large retraction
- use fast retraction speed

And then once the above settings were at a point where blobbing started to be reduced, I was finally able to see differing results when testing pressure advance.

Before going further, I must state that the acceleration speeds I ended up needing, and have listed below, are not typical for a Tronxy x5s and are quite high in comparison to the speeds in the default config for the Tronxy x5s, and are only listed below for discussion and feedback purposes. If you decide to use these settings, you do so of your own free will and you accept all risks.

Klipper printer.cfg (adjust/add under section listed) I am currently testing:
[extruder]
max_extrude_only_velocity: 250
max_extrude_only_accel: 9000
pressure_advance_lookahead_time: 0.01
pressure_advance: 0.15
[printer]
max_velocity: 300
max_accel: 6000

Cura 3.4.1 retraction and travel settings I am currently testing:
Retraction Distance: 10mm
Retraction Speed: 250mm/s
Travel Speed: 200mm/s

There is still a slight pause at the start of each layer and/or line segment, but increasing the acceleration seems to have reduced the amount of time it pauses for, and in turn helped to reduce the blobs.

However, it comes at a compromise, as the increase in acceleration introduces other problems due to the greatly increased movement speed - such as vibration.

To be honest, I'm still not convinved this isn't an Octoprint/ControlBoard/Klipper/serial problem, as it doesn't pause like that when using Repetier. But the speed increase available when using Klipper is such a plus that I'm now trying to decide if the compromise is worth it.

Ideally I'd be able to reduce the acceleration back to what is in the default printer config and come up with a different solution to fix the blobbing, but I'm lost on what to try.

Any suggestions and comments that might help are appreciated.

KevinOConnor, 2018-08-21:
@unnefer1 - interesting.  I'm aware of two general ways to reduce the impact of pressure: use pressure advance, avoid pausing the head.  It looks like you've gone the second route.  That route has some drawbacks (namely ringing and possibility of lost steps).

Unfortunately, the two routes are generally mutually exclusive - with pressure advance, you actually want somewhat longer cornering times so that there is sufficient time to retract the filament and eliminate the pressure.  In contrast, when avoiding pauses, you want a very high acceleration and high square_corner_velocity.

It would help if you could print out some of the squares documented in https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md and post pictures of them.  It's hard to give advice when we can't see what you're seeing with a known test.  The square.stl file is just a standard stl file (you can find it in the klipper/docs/prints/ directory on github or on the machine running klipper).  For a bowden setup, I'd run the square tests with square_corner_velocity=0 and max_accel=500.

It's also possible the extruder issues you are seeing are related to something else - filament temperature, excessive moisture in the filament, something odd that the slicer is doing, z offset issues, etc.

-Kevin

unnefer1, 2018-08-21:
Hi @KevinOConnor. For whatever reason, when I download the "square.stl" file from the pressure advancedoc, the stl will not load in Cura - Cura just gives an error?

Are you able to zip the stl and include a direct link to the zip file instead?

The next real chance I'll get to do further testing is on the weekend again. If you have a moment spare  and can provide the direct link, I'll tests again using stock config and pa from 0.5 through to 0.8, as well as acceleration from 100 through to 5000 and upload photos of each resulting printing.

KevinOConnor, 2018-08-21:
> The next real chance I'll get to do further testing is on the weekend again. If you have a moment spare and can provide the direct link, I'll tests again using stock config and pa from 0.5 through to 0.8, as well as xy acceleration from 500 through to 5000 and extruder acceleration from 1000 through to 9000 upload photos of each resulting printing.

Great.  I recommend you run the square tests at max_accel=500, square_corner_velocity=0, pressure_advance values from 0.05 up to 1.0, extrude_only settings at their defaults, disable retraction in the slicer.  I suspect varying the acceleration will just complicate the test.

-Kevin

cfelicio, 2018-08-29:
I'm also curious to see the results. My direct driver printer saw a huge improvement in quality with Klipper (wanhao i3), once everything was tuned. Funny enough, before I was not able to use S3D to print with the i3, it was always a hit or miss. The stock firmware was prob garbage.

On my D Bot with a 60cm bowden, I'm having pressure issues. My stock S3D settings end up in pressure buildup on the bowden tube, and that results in blobs, and the occasional extruder skipping.

My E-steps are calibrated, and doing the 1 wall test cube results in a 0.90 extrusion multiplier, but if I use it, I end up over extruding. Using 0.80 works well, but I can see some under extrusion between perimeters.

What I will try to do next is move the extruder close to the hotend, and hopefully shorten the bowden a little bit. From what I read, this should help. :-)

closed 2018-09-06 by KevinOConnor

cfelicio, 2018-10-22:
Update on this, in case anyone comes across this post. I tried a shorter bowden (around 40cm), but still quality was subpar when compared to direct drive. I ended up converting my printer to Direct Drive, and couldn't be happier. One of the drawbacks of having Direct Drive is the extra load on the carriage, but I did not notice any new ringing / artifacts. On the other hand, retraction can be now 1mm, and it works perfect.

antst, 2019-10-27:
I am fighting similar issues. Direct drive, vocano, petg, 0.6 nozzle.
Pressure advance is tuned. I also tried with pressure advance smoothing PR, it works in other aspects, but this one it doesn’t change.  All “stops” result in blobbing and stringing, as those pauses are enough to get some leaking. This is volcano, so we have quite some amount of melted plastic and gravity...pauses aren’t very compatible with volcano. And retract or pressure advance can do little: once we stop. It leaks. With volcano you need to always move. At least with petg.

On the same setup I have nearly zero stringing and totally zero blobbing with marlin (also using pressure/linear advance).

I can try to reduce “stops” by lifting retraction speed and acceleration, but this isn’t good way, moreover with geared extruder acceleration limit is lower, plus pancake stepper will start skipping very quickly.