Klipper3d/klipper #518: The engine braking mode is needed

AlexSkarbo, 2018-08-05:
I would like to have the opportunity, as in smoothies, that the motors would slow down the movement along the axes, this is very important for the Z-axis, especially on the belts

AlexSkarbo, 2018-08-06:
I will answer rudely, but do not be offended, I do not want to offend anyone, but I think it's not worth explaining to those who are off topic, or do not understand what I'm talking about. I do not know how this is easier to explain, but I need to have a braking mode, for example restoration, if the engines do not have power. I can show video how smoothies work and how the clipper works.

dragonnn, 2018-08-06:
If they is no power, how on earth should be software able to do anything? 
You need a hardware solution, not software.

AlexSkarbo, 2018-08-06:
Eh ... I do not have to explain that the motor can be a current generator ...

AlexSkarbo, 2018-08-06:
https://drive.google.com/file/d/1kosWhhOgSBV6S9Ptv7E7ia2pY2uKITja/view?usp=sharing Smoothies without power

AlexSkarbo, 2018-08-06:
https://drive.google.com/open?id=1Xt3i1oS7RePCzEeJqJrm87-d4u1KGII2 On a printer with klipper

AlexSkarbo, 2018-08-06:
https://en.wikipedia.org/wiki/Regenerative_brake

jakep82, 2018-08-06:
> In the end, could you please, tell me, what should I do, that it would be impossible to move the engines, in the absence of energy?

This is a hardware problem, not a software problem.  Klipper isn't running when the printer isn't powered.

AlexSkarbo, 2018-08-06:
I still try to explain that if there is no current in the common power network, then the motor, if it is moving, generates energy as power generator. If anyone does not know, then it has been used for a long time in electric cars and etc machines, more than 100 years ago. I have not gone into the specifics of implementation, but I think that should not be problem. Probably just did not come across at this problem. But this is very important at now.

AlexSkarbo, 2018-08-06:
I think it's not worth explaining why it's important? Simply there can be a lot of use cases, for example very rarely but lack of electricity, much more often, replacement of the filament and simply disabling the engines, if not working, after it is very difficult to restore the print

AlexSkarbo, 2018-08-07:
As I understand it, the klipper consists of two parts, the first works on microcomputer, and the other on a microcontroller, to which the motors are connected via drivers, this is my question for the second part ...

AlexSkarbo, 2018-08-07:
I would call this mode self-braking

dragonnn, 2018-08-07:
I am sure most people here know what regenerative breaking is and that a motor when turning can works as a generator. So don't be rude.
And still, this is not possible, the power generated by a motor is most time not enough to power the mCPU up. 
Second - you "Smoothies"  video, I think you mean smoothieware running on a printer - I doesn't have regenerative breaking too. This what you shows on the video is just how CoreXY kinematics works - due the belts configuration it have much more resistance when moving so it looks like regenerative breaking but this is not. I know what about I am talking since I have a CoreXY printer too and it have exactly the same resistance when moving (even when I physically disconnect the motors).

The only way to really do a regenerative breaking is using a normal closed relay with shorts the motor coils. And when the software boots up it can open they relay. But this has nothing to do with software, no way software can do regenerative breaking itself using only the power with is generated by moving the motor.

AxMod3DPrint, 2018-08-07:
As @dragonnn says what you want implementing _*in software*_ is something that is a _*hardware*_ issue. I'm really not sure what you were fiddling with at the back of the CoreXY, to release the motors, but having both Smoothie and Klipper printers, I think I know what you're getting at. 

Smoothie by default when on will keep the motors enabled and has no time out, while Klipper (and Marlin) do not and have a time out. Klipper's time out can be increased in the configuration file via this section:

`# Idle timeout. An idle timeout is automatically enabled - add an`
`# explicit idle_timeout config section to change the default settings.`
`[idle_timeout]`
`gcode:`
`#   A list of G-Code commands (one per line) to execute on an idle`
`#   timeout. The default is to run "M84".`
`timeout: 600`
`#   Idle time (in seconds) to wait before running the above G-Code`
`#   commands. The default is 600 seconds.`

If you want the motors to be on as soon as you switch power on to the printer, it can be done via a script, say on connecting Octoprint to the printer or whatever host software you are using.

megablue, 2018-08-07:
> I will answer rudely, but do not be offended, I do not want to offend anyone, but I think it's not worth explaining to those who are off topic, or do not understand what I'm talking about. I do not [...]

How can you expect someone to understand what you mean by an extremely vague request? 

As others have mentioned, you need a hardware solution, not a firmware one. there is no way that Klipper can do anything regarding braking with the stepper motors without hardware changes.

KevinOConnor, 2018-08-16:
It's not clear to me what is being requested.  In any case, I suspect you will need to implement it should there be development work necessary.

-Kevin

closed 2018-08-16 by KevinOConnor

myk3y, 2018-11-27:
If I read it right, he wants to implement some sort of handbrake for when the printer is off.

Easy!

Make a switch that when open routes the stepper power normally from the driver/ocntroller, when engaged crosses the two windings, as you would do when trying to work out which wires are for which windings and you get increased resistance. It can be manual, or via a solenoid triggered by a pin on the board.