Klipper3d/klipper #519: Extrude below minimum temp on Duel X Carriage printer using T1 only to print

bruce356, 2018-08-05:
When using Tool-1 only, on the printer, just before printing is to start, Klipper  stops with an error "Extrude below minimum temp". Both "min_extrude_temp" are set to 170 deg. 
To overcome this problem I have to set "min_extrude_temp" on Tool-0 in printer.cfg to something below room Temp (usualy I just remove  the zero so 170 deg.becomes 17 deg.).
[Klippy Log & Octopi.zip](https://github.com/KevinOConnor/klipper/files/2260070/Klippy.Log.Octopi.zip)
Thanks & regards - bruce

KevinOConnor, 2018-08-06:
As mentioned in issue #510 - you need to reproduce the problem running code from this repository before open a github issue in this repository.

bruce356, 2018-08-07:
@KevinOConnor Hi being new I was not aware of the rule and I did not understand what you meant re the comment on#510,sorry.
I have now a fresh install of your latest Master branch. The problem is the same as first stated.
Latest Klippy Log & Octopi Terminal
[Klippy Log & Octopi_1.zip](https://github.com/KevinOConnor/klipper/files/2266784/Klippy.Log.Octopi_1.zip)
Thanks & regards - bruce

KevinOConnor, 2018-08-07:
In order to get more debugging information in the log, you should issue an M112 command immediately after the error.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

The code should only be checking for the active extruder.  Make sure you've issue a T1 command prior to any extruder movement if you only want to use extruder1.

-Kevin

bruce356, 2018-08-08:
@KevinOConnor , please find attached the required info this time also including Gcode.
Regards - bruce
[Klippy Log & Octopi_1+Gcode.zip](https://github.com/KevinOConnor/klipper/files/2268549/Klippy.Log.Octopi_1%2BGcode.zip)

KevinOConnor, 2018-08-09:
The log indicates the following commands were sent to Klipper:
```
T0
G1 X-41 F10000
G1 Y160 F10000
M104 S210 T1
M109 S210 T1
G92 E0
G1 E-0.8000 F2100
```

The first line (T0) activates extruder0 and the last line is a request to retract on extruder 0.  That's not valid if extruder 0 has not been heated.  A "T1" command should be used to activate extruder 1 prior to any extrusion or retraction commands if you only want to use extruder 1.

-Kevin

bruce356, 2018-08-10:
Hi @KevinOConnor, thanks for the reply and solution.
Those last entries are not in my start script and are placed into the Gcode by Simplify 3D (pain).
I have never taken much notice of them in the past as they were never a problem in Marlin printing with T1 only.
If a tool is not up to minimum temperature in Marlin that extruder will simply not feed any filament, but does not cause an error and stop the print (so it was of no consequence).

Obviously Klipper does things differently.
Regards - bruce

closed 2018-08-14 by bruce356