Klipper3d/klipper #520: [heater_fan fan1] problem on Dual X Carriage printer RUMBA board

bruce356, 2018-08-05:
Hi, Printer has two independent hot ends heated separately, printing can be done with both hot ends or individually as you know. Problem is the  [heater_fan fan1] comes off the same Pin ar8 for both fans (one fan for each hot end E3D's v6). So if I configure:-
[heater_fan fan1]
pin: ar8
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
then the fans activate if I am printing with T0, but if printing with T1 the fans do not come ON and visa versa if configured as below:-
[heater_fan fan1]
pin: ar8
heater: extruder1
heater_temp: 50.0
fan_speed: 1.0

Hoping you have a solution to this
regards - bruce
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2260078/klippy.zip)

unnefer1, 2018-08-06:
Have you tried only setting the pin and letting Klipper defaults decide the rest?

[heater_fan nozzle_cooling_fan]
pin: ar8

bruce356, 2018-08-06:
@unnefer1 thanks for the reply, this only functions with T0, when T1 is heated the fans do not turn on.
regards

KevinOConnor, 2018-08-06:
It would require a code change to implement that.  I believe @gdachs had some test code for it.

-Kevin

bruce356, 2018-08-10:
@gdachs or @KevinOConnor , I have installed gdachs version of clipper but have had no luck in finding or figuring out what Config is required to make the "Multiple extruders using single fan Pin" work.
Any help would be appreciated 
Thanks and regards - bruce

gdachs, 2018-08-10:
@bruce356 I am on vacation, 4 hours flight away from everything. My printer is work in progress. The only thing it can currently do is homing X and Y. So I will not be able to find the issue anytime soon.

bruce356, 2018-08-11:
@gdachs, thanks for the reply (sorry did not know you are on holiday), KevinOConnor mentioned 5 days ago that you had made some code changes to Klipper to enable Heater fan1 from one pin (in my case ar8) to turn on, when either of two independent hot ends are used (in other words I print with T0 only or with T1 only).

With your changed version of Klipper what configuration in "Printer.cfg" is used/required in order for the heater fan1 to turn on when heating with either T0 or T1 for printing.

I hope you understand what I am trying to explain.
thanks again, regards - bruce

klHaribo, 2018-08-13:
I have the same problem: printer is configured with a chimera. If I setup two fans(for one pysically) with the same pin for the heater_fan, klipper reports that the pin is already configured. I thing we need something like this:

[heater_fan fan1]
pin: ar8
heater: extruder

[heater_fan fan2]
shared_pin: fan1  # in this way you can print only with second extruder (ie. desired color is loaded)
heater: extruder1

bruce356, 2018-08-14:
@KevinOConnor Hi I have tried @gdachs version of Klipper, as it is it fails to connect to my printer, "communication error". I do not expect you to diagnose the problem (obviously) as this is not your official version, but I was hoping you might find the time to code something that would work and after testing then include it in the master version. I assume there will be more users in future with this problem on IDEX printers.
Thanks & regards - bruce

gdachs, 2018-08-14:
@bruce356 If you don't want to wait until I made it ready, then it makes no sense to forward that last comment to me.

bruce356, 2018-08-15:
@gdachs sorry my mistake

KevinOConnor, 2018-08-17:
I added support for this (commit 1b07be07).  With this update, one can now define multiple heaters for a heater_fan.  For example:
```
[heater_fan my_cooling_fan]
pin: ar8
heater: extruder, extruder1
```

-Kevin

gdachs, 2018-08-17:
Looks a little bit like my implementation. But I think I forgot to initialize the array.

bruce356, 2018-08-18:
@KevinOConnor , thank you very much it functions perfectly
regards - bruce

closed 2018-08-20 by bruce356