Klipper3d/klipper #521: cannot flash Anet A6

jesperldk, 2018-08-05:
I am tying to flash Klipper to my Anet A6 (which has the same board as an A8) from my Zero. I am using the command I got from the printer-anet-a8-2017.cfg file, but I get "not in sync" errors.
The Anet is a bit picky to flash, also when I flash Marlin.
Does anybody have any suggestions? When I flash Marlin, I do it from the Arduino App on my pc - with the right cable that works. Is there a recipe on how to do it that way with Klipper?

jesperldk, 2018-08-05:
Sorry for the trouble - I got it now. Changed to default baud rate, and that made it flash. It sure is picky...

closed 2018-08-05 by jesperldk

MWEVERETT, 2018-08-05:
Flash the Optiboot bootloader. and try that one.

Mrk

jesperldk, 2018-08-05:
Already had Optiboot on it ;-) Anyway, different baud did it...