Klipper3d/klipper #523: M117 gcode doesn't work when the message starts with a digit.

raparada, 2018-08-06:
M117 gcode doesn't work when the message starts with a digit. That's the case of ipOnConnect octopring plugin. Same issue if I try manually: "M117 192.168.x.x" 
To solve this I had to modify the plugin's code by adding a "_" before server_ip.

Thanks!!

KevinOConnor, 2018-08-16:
Indeed it does not work.  This doesn't look simple to solve though, because of the funky g-code parsing rules.  I'll try to take a look at it, but it's not a high priority for me right now.

-Kevin

raparada, 2018-08-17:
Thanks you Kevin for you answer and you fantastic work!!

imrcly, 2018-08-21:
Just a note it doesn't like [ either

KevinOConnor, 2018-08-22:
This should be fixed now (commit e94f3b78).

-Kevin

closed 2018-08-30 by KevinOConnor

raparada, 2018-08-30:
Thank you all!!
-Rodrigo