Klipper3d/klipper #529: [Q]Multiple pin outputs...HOWTO

Hywelmartin, 2018-08-08:
Hi everybody

First of all Great work

I can't find any more info on this feature than what is in the  klipper/config/example-extras.cfg...

As I'm not really familiar with coding stuff this is hard for me to understand..

# Multiple pin outputs (one may define any number of sections with a
# "multi_pin" prefix). A multi_pin output creates an internal pin
# alias that can modify multiple output pins each time the alias pin
# is set. For example, one could define a "[multi_pin my_fan]" object
# containing two pins and then set "pin=multi_pin:my_fan" in the
# "[fan]" section - on each fan change both output pins would be
# updated. These aliases may not be used with stepper motor pins.
#[multi_pin my_multi_pin]
#pins:
#   A comma separated list of pins associated with this alias. This
#   parameter must be provided.

enable_pin: !ar56

45 and 7

My needs...

While the stepper drivers are engaged(enable_pin: !ar38) I would like to activate two more pins 45 and 7

[minikosseldeltaklipper.txt](https://github.com/KevinOConnor/klipper/files/2270572/minikosseldeltaklipper.txt)

FHeilmann, 2018-08-09:
```[stepper_x]
enable_pin: multi_pin:my_enable_pin

[multi_pin my_enable_pin]
pins: !ar38, ar45, ar7```

Hywelmartin, 2018-08-09:
thanks, I will try it..

Pin 7 is for a steppper driver coolingfan so..