Klipper3d/klipper #530: Can't use dual Z stepper configuration

ich777, 2018-08-10:
Hello,

i can't get dual Z steppers to work with my CoreXY.
If i configure all and send a "firmware_restart" to my printer this error appears:

> Recv: // Error loading kinematics 'corexy'
> Recv: // Once the underlying issue is corrected, use the "RESTART"
> Recv: // command to reload the config and restart the host software.
> Recv: // Printer is halted
> Recv: !! Error loading kinematics 'corexy'

Configured it as discribed in the documentation.

Regards

sniperlucian, 2018-08-10:
please don't forget to attach the log file. (/tmp/klippy.log) 

its mostly very small config errors - no way to help you with that little information .... I have three motor Z axis and needed to twiddle with config too.

StefanMeGit, 2018-08-10:
I guess I have to use it like this:
stepper_z = first z Stepper 
stepper_z1 = second z Stepper
...

U always have to start with “stepper_z”

radry, 2018-08-12:
Can confirm the issue. CoreXY can't be used when having more than one Z stepper configured.

wizhippo, 2018-08-13:
@radry It works fine for me but I use separate end stops.  You have the same end stop pins being used for both steppers.  How is it to know what stepper it is that activated the end stop?   Pretty sure you will need to have different end stop pins on each.

sniperlucian, 2018-08-13:
i have 3 z stepper with one Z probe and it works. If you want to use the same endstop at one input pin try configuration stated in the docs, namely comment out the endstop for stepper_z1:

#endstop_pin: ^ar19
#   If an endstop_pin is defined for the additional stepper then the
#   stepper will home until the endstop is triggered. Otherwise, the
#   endstop will home until the endstop on the primary stepper for the
#   axis is triggered.

ich777, 2018-08-13:
@wizhippo & @sniperlucian can you post the section where you configured the z steppers of the printer.cfg, would be great to see where the problem in my configuration is.

radry, 2018-08-13:
@wizhippo @sniperlucian 
Thank you. but How would I use that with a probe? There is a seperate probe configuration in the config, but that would also use the same pin as the endstop since the probe is the endstop.  Should I delete the endstop from both Z steppers in that case or can probe and endstop use the same pin?

ich777, 2018-08-13:
Problem solved, i must name the steppers: [stepper_z] & [stepper_z1] not [stepper_z1] & [stepper_z2]...
Thank you everyone ;)

KevinOConnor, 2018-08-16:
@radry - if you want to use a Z probe to stop all Z steppers, then you would send the stepper_z endstop_pin to "probe:z_virtual_endstop" and not set the endstop_pin in any other Z stepper section.

-Kevin

closed 2018-08-16 by KevinOConnor

radry, 2018-08-19:
@KevinOConnor 
Sorry that doesn't work:  
I set the stepper_z endstop_pin to "probe:z_virtual_endstop" and it gives following error:

> Recv: !! Option 'position_endstop' is not valid in section 'stepper_z1'

```
[stepper_z] 
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .000625
position_endstop: -0.91
position_max: 200
position_min: -1

[stepper_z1]
step_pin: ar36
dir_pin: ar34
enable_pin: !ar30
step_distance: .000625
endstop_pin: probe:z_virtual_endstop
position_endstop: -0.91
position_max: 200
position_min: -1

[probe]
pin: ^ar18
#   Probe detection pin. This parameter must be provided.
z_offset:-0.91
#   The distance (in mm) between the bed and the nozzle when the probe
#   triggers. This parameter must be provided.
speed: 5.0
#   Speed (in mm/s) of the Z axis when probing. The default is 5mm/s.
```

AxMod3DPrint, 2018-08-19:
Remove the position_endstop  from stepper_z1.

'Eagles may soar, but Weasels don't get sucked into Jet engines'