Klipper3d/klipper #532: src/sam3x8e/Makefile:33: recipe for target 'flash' failed

StefanMeGit, 2018-08-10:
Since a few days, I cannot flash my Arduino due... I always get this message:
```
make flash FLASH_DEVICE=/dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_55730323831351418181-if00
  Flashing out/klipper.bin to /dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_55730323831351418181-if00 via bossac
No device found on serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_55730323831351418181-if00
src/sam3x8e/Makefile:33: recipe for target 'flash' failed
make: *** [flash] Error 1
```
But Klipper is still running. I can connect normally and print well... just the FLASH process is not working for me. Any ideas?

I guess the KLIPPY log is not responsible for this problem. What log file do I have to upload?

khseal, 2018-08-11:
Try to /dev/ttyUSB0 or  /dev/ttyACM0

StefanMeGit, 2018-08-11:
I already did. No difference...

StefanMeGit, 2018-08-13:
I am still not able to flash my due. I guess its a problem by the due. I ordered one form AliExpress and will close this issue after I replaced it with success.

khseal, 2018-08-13:
Try flash in Arduino IDE, or flash in other computer.

KevinOConnor, 2018-08-16:
On the Due, one can not use `make flash FLASH_DEVICE=/dev/serial/by-id/...` .  One would have to find out what the raw serial id is and then flash using something like `make flash FLASH_DEVICE=/dev/ttyUSB99` .  It is still okay to use the /dev/serial/by-id/ form in the printer.cfg file.

-Kevin

StefanMeGit, 2018-08-17:
No u missundetood me. It works with the `make flash FLASH_DEVICE=/dev/serial/by-id/...` fine. I run the the due with this configuration. I Just coulnt flash it anymore... with either of make flash `FLASH_DEVICE=/dev/serial/by-id/...` or `FLASH_DEVICE=/dev/ttyUSB99` maybe just my due is broken. I can try out something to flash with MALRIN or something lese,  because is is my only DUE I have running. So I ll go on with this situation now, until my other DUE is arriving

StefanMeGit, 2018-08-22:
The new DUE works just as expected. Was an issue with the DUE. 

thanks.

closed 2018-08-22 by StefanMeGit