Klipper3d/klipper #534: Internal error on comand: "G1"

ChiliApple, 2018-08-11:
hy,

I had the error on 2 prints, the GCode seems to tune, but something about the conversion of the steps seems to be flawed ?!

in my config I have this value for my stepper "step_distance: .00234665"
klipper is up to date ...

[image]

AxMod3DPrint, 2018-08-13:
Please attach your klippy.log to this issue as this will give more information.

ChiliApple, 2018-08-13:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2283894/klippy.log)

gdachs, 2018-08-13:
There is an unbelievable big amount of extrusion. This gcode line looks fishy.

ChiliApple, 2018-08-13:
the gcode looks ok in my opinion, but the value in the terminal not

Arksine, 2018-08-13:
As long as that is absolute extrusion I don't see a problem with it.  The asterisk you see in the terminal is just a checksum, so the terminal gcode is okay too.

The klippy.log you attached doesn't seem to show this error, it just shows where you entered M112 to shut down the system.  Can you reproduce the error and attach the accompanying klippy.log?

KevinOConnor, 2018-08-23:
There isn't much we can do without a log file from the event itself.  I'm going to close this - I'll reopen if there is a log file.

closed 2018-08-23 by KevinOConnor