Klipper3d/klipper #536: Error loading kinematics

tuwwe, 2018-08-13:


gdachs, 2018-08-13:
Why not first fix the missing endstop pins in the config file that are mentioned in the log? Maybe the error is then gone.

tuwwe, 2018-08-13:
here is printer.cfg file and they are there 
i dont know why ist show them log file

# OULWare PrinterController v1.0 + DUE
#    - TMC2130 on XY axis!
#    - DRV8825 for Z and E0 & E1
#
#  === DRIVERS ===
#
#  -------------------
#  | X | End MIN | D54
#  | X | End MAX | D55
#  | Y | End MIN | D56
#  | Y | End MAX | D57
#  | Z | End MIN | D58
#  | Z | End MAX | D59
#  -------------------
#
# === HEATERS ===
# | AUX | D9 |
# | BED | D8 |
# | E0  | D7 |
# | E1  | D6 |
#
# === THERMISTORS ===
# | T0    | A9  |
# | T1    | A10 |
# | T2    | A11 |
# | TC CS | D47 |
#
# === FAN ===
# | FAN0 | D10 |
# | FAN1 | D11 |
# | FAN2 | D12 |
# | FAN3 | D13 |
#
# === STEPPER X ===
# | STEP | D38 |
# | DIR  | D36 |
# | EN   | D40 |
# | CS   | D34 |
#
# === STEPPER Y ===
# | STEP | D28 |
# | DIR  | D26 |
# | EN   | D30 |
# | CS   | D32 |
#
# === STEPPER Z ===
# | STEP | D23 |
# | DIR  | D29 |
# | EN   | D25 |
# | CS   | D27 |
#
# === STEPPER E1 ===
# | STEP | D44 |
# | DIR  | D42 |
# | EN   | D46 |
# | CS   | D48 |
#
# === STEPPER E2 ===
# | STEP | D33 |
# | DIR  | D31 |
# | EN   | D35 |
# | CS   | D37 |
#
# === STEPPER E3 ===
# | STEP | D41 |
# | DIR  | D39 |
# | EN   | D43 |
[... 342 lines cut ...]
max_z_accel:    30
motor_off_time: 600
junction_deviation: 0.2 ; default 0.02
require_home_after_motor_off: false
allow_move_without_home:      true
;sw_limit_check_enabled:       false

[idle_position]
idle_position_x:		10.0
idle_position_y:		10.0
idle_position_z_lift:		5.0
idle_position_travel_speed:	2000

[gcodes.arch]

[gcodes.retract]
test = false

[virtual_sdcard]
path: ~/.octoprint/uploads/

[hostcmd]
commands: 
  list = ls -al
  echo = echo "this is a test"

jakep82, 2018-08-13:
This a config file for the Cruwaller/OulWare fork of Klipper.  It's heavily modified from mainline Klipper so you aren't going to be able to get support here.

tuwwe, 2018-08-14:
Thats i have oulware board and here is pins easy  modify but it doesnt matter what config i put it says same problem

jakep82, 2018-08-14:
If you installed from this Git repository you need to create your own custom config file.  I have an OULWare board and that's what I did.  The config file your using will only work if you installed from the here [https://github.com/OULWare/klipper](url).

obelisk79, 2018-08-15:
definitely an issue with your config file. looks like all of the section headers are non-standard and that is most likely the source of your kinematics error. I got the exact same error when I tried to organize the config file for better readability using tab characters (aka unexpected things in the config text).

gdachs, 2018-08-15:
@obelisk79: the error message claims: `NoOptionError: No option 'endstop_pin' in section: 'stepper_x'`
and he has really no option "endstop_pin" in the config. So why to look for other reasons?

obelisk79, 2018-08-17:
@gdachs If you look where he re-posted his config a second time, the endstop pin appears to be defined. so I offered, based on my own experience that it may be a formatting/syntax error.

tuwwe, 2018-08-18:
I have been fighting with config file, and always come other error , tryed others working configs what have done to oulware board but doesnt work on my klipper, i dont try drive stepper only tryed status and then i get error

tuwwe, 2018-08-18:
=============== Log rollover at Sat Aug 18 13:37:43 2018 ===============
INFO     root :: Starting Klippy...
INFO     root :: Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.5.0-423-g899b180'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
INFO     printer.gcode :: queue fileno <Queue.PriorityQueue instance at 0x75a47210>
INFO     printer :: Start printer at Sat Aug 18 13:37:44 2018 (1534588664.7 7.4)
INFO     config :: ===== Config file =====
[driver drv8825_x]
type = DRV8825
microsteps = 32

[driver drv8825_y]
type = DRV8825
microsteps = 32

[driver tmc2130_x]
type = TMC2130
ss_pin = ar34
spi_mode = 3
spi_speed = 8000000
current = 1000.0
microsteps = 32
mode = stealthChop
diag0_out = temp_prewarn
diag1_out = stall
diag0_active_high = 1
diag1_active_high = 1
interpolate = 0
sensor_less_homing = 1
stealth_max_speed = 100
stall_threshold = 3

[driver tmc2130_y]
type = TMC2130
ss_pin = ar32
spi_mode = 3
spi_speed = 8000000
current = 800.0
microsteps = 32
mode = stealthChop
diag0_out = temp_prewarn
diag1_out = stall
diag0_active_high = 1
diag1_active_high = 1
interpolate = 1
sensor_less_homing = 1
stealth_max_speed = 100
stall_threshold = 10

[stepper_x]
driver = tmc2130_x
motor_step_angle = 1.8
pitch = 2.
teeths = 16.
step_pin = ar38
dir_pin = ar36
enable_pin = ar40
endstop_min_pin = ar54
endstop_max_pin = ar55
position_min = 0
position_max = 220
position_endstop = 0
homing_offset = 0
homing_speed = 40
homing_direction = min
homing_retract_dist = 0

[stepper_y]
[... 180 lines cut ...]
  File "/home/pi/klipper/klippy/klippy.py", line 251, in _try_load_extensions
    init_func(self, config)
  File "/home/pi/klipper/klippy/gcodes/gui_stats.py", line 395, in load_gcode
    GuiStats(printer, config)
  File "/home/pi/klipper/klippy/gcodes/gui_stats.py", line 16, in __init__
    self.babysteps = printer.lookup_object('babysteps')
  File "/home/pi/klipper/klippy/klippy.py", line 202, in lookup_object
    raise self.config_error("Unknown config object '%s'" % (name,))
Error: Unknown config object 'babysteps'
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 303, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 273, in _read_config
    self._try_load_extensions('gcodes', 'load_gcode', config)
  File "/home/pi/klipper/klippy/klippy.py", line 251, in _try_load_extensions
    init_func(self, config)
  File "/home/pi/klipper/klippy/gcodes/gui_stats.py", line 395, in load_gcode
    GuiStats(printer, config)
  File "/home/pi/klipper/klippy/gcodes/gui_stats.py", line 16, in __init__
    self.babysteps = printer.lookup_object('babysteps')
  File "/home/pi/klipper/klippy/klippy.py", line 202, in lookup_object
    raise self.config_error("Unknown config object '%s'" % (name,))
Error: Unknown config object 'babysteps'

this is what i get now

AxMod3DPrint, 2018-08-18:
Unfortunately, as you are using the OULWare fork, we really cannot help here. This is throwing an error regarding babystepping `Error: Unknown config object 'babysteps'` which is something that is not included in the Master of Klipper. You are better off posting on the OULWare fork for help with this.

gdachs, 2018-08-18:
@tuwwe learn to attach the logfile instead of pasting it to the message.
Anyway, as you can see my advice did help. You get now another error, I have only no idea why you don't see it. 
It seems rather obvious, it says: "Error: Unknown config object 'babysteps'"
So remove it from the config.

tuwwe, 2018-08-18:
Srry about that, im new here and dont know all tricks here, i would remove it from config but i dont have it in config, only babysteps.py in gcodes folder

obelisk79, 2018-08-18:
Post your config in pastebin

gdachs, 2018-08-18:
@tuwwe: sorry you are right, I misread the error message. As you got told now multiple times, you are using a fork of klipper that is incompatible with this from KevinOConnor. You should ask the owner of this fork for support.

closed 2018-08-19 by tuwwe