Klipper3d/klipper #537: Filament runout sensor?

MrMbss1, 2018-08-13:
Is there no implementation of a filament runout feature in Klipper? I feel like it's a fairly simple and and necessary function to have

Hywelmartin, 2018-08-15:
Then M600 (Filament change) as well

KevinOConnor, 2018-08-16:
This looks like a duplicate of issue #249 .

mamoit, 2018-08-17:
Efforts on #545, ideas and requests are welcome.

AxMod3DPrint, 2018-08-21:
A note on M600, technically, if using octoprint, if the filament runs out, it could just call a `@pause` and the park movement can be done from within octoprint from the GCode Scripts tab like this:

_After Pause_
```
{% if pause_position.x is not none %}
; relative XYZE
G91
; retract filament of 0.8 mm up, move Z slightly upwards and 
G1 Z+5 E-0.8 F4500
; absolute XYZE
G90
; move to a safe rest position, adjust as necessary
G1 X0 Y0
{% endif %}
```
_Before resume_
```
{% if pause_position.x is not none %}
; relative extruder
G91
; prime nozzle
G1 F4500
G1 E-0.8 F4500
G1 E0.8 F4500
G1 E0.8 F4500
; absolute E
; absolute XYZ
G90
; reset E
G92 E{{ pause_position.e }}
; move back to pause position XYZ
G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F4500
; reset to feed rate before pause if available
{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}
{% endif %}
```

A quick note on the `G1 F4500` after the prime, this is a work around for a known bug where extruder only moves cause a problem with `max_extrude_cross_section`

KevinOConnor, 2018-09-25:
Closing in favour of existing #249.

closed 2018-09-25 by KevinOConnor