Klipper3d/klipper #539: Move exceeds maximum extrusion - Error

gennro, 2018-08-14:
Alright searched through the issues and couldn't find a solution to this problem.

Machine is a corexy running two MCUs one LPC1768 ASZMS Mini (runs all axes) and a AVR ATMEGA2560 MKS Gen L (runs extruders, heaters, temps sensors). I am using a quad filament dual nozzle Chimera hotend with dual y splitters.

Alright at about layer 5 to 6 I get the Move exceeds maximum extrusion (148945865169661280.000mm^2 vs 6.000mm^2) during my filament switching gcode which is this so this means this code works fine for a few times before this.

T[old_tool]
G1 E-4  F5000
G1  Y10 F15000
G4 P1000
G1 E3.9 F4000
G1 E-20 F1200
G1 E20.5 F4000
G1 E-120 F4000
T[new_tool]
G1 E110 F2000
G1 E3 F300

I have removed all coordinate switching commands that I could find to make sure this wasn't an issue.
Also does it on different models.
I can print with one filament just fine, but when I go to do multiple materials it does this error.

I went through all the logs/gcode to see if anything was causing this and couldn't find anything.

Also I have this same gcode running a a smoothieware based board just fine.

I have included all relevant information below that may be helpful.

[printer config.log](https://github.com/KevinOConnor/klipper/files/2288137/printer.config.log)
[Moai_Body gcode .log](https://github.com/KevinOConnor/klipper/files/2288132/Moai_Body.gcode.log)
[klippy.log](https://github.com/KevinOConnor/klipper/files/2288131/klippy.log)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/2288130/octoprint.log)

https://i.imgur.com/R4gaMvH.jpg

AxMod3DPrint, 2018-08-15:
Looks like your `max_extrude_cross_section` is to blame here. Test with it commented out  to check. For your setting it should be 6*(6^2) = 216. (The default is: 4.0 * nozzle_diameter^2)

gennro, 2018-08-15:
I left it on default and changed it to various numbers and nothing would fix it, I wish I could disable to the max extrude cross section check to see what happens. @AxMod3DPrint

KevinOConnor, 2018-08-16:
Alas, it appears you've hit the bug described in issue #461.  Until this bug can be fixed, the workaround is to issue a dummy move without extrusion (eg, `G1 F15000`) after a tool change and prior to an extrude only move.

-Kevin

gennro, 2018-08-16:
Ah thank you Kevin, I wasn't sure if the bed tilt issue was the same as this, I will give this a try then, I have ran into the bed tilt issue a lot.

gennro, 2018-08-16:
That looks like it fixes it. So basically anyone that has issues with this do like Kevin says put a dummy move right before an extrusion move and it will fix the problem.

T[old_tool]

G1 E-4  F5000

G91
G1 Z2 F300
G90
G1  Y20 X20 F18000

G4 P1000

G1 F15000
G1 E3.9 F4000
G1 F15000
G1 E-20 F1200
G1 F15000
G1 E20.5 F4000
G1 F15000
G1 E-120 F4000

T[new_tool]

G1 F15000
G1 E110 F2000
G1 F15000
G1 E3 F300

G91
G1 Z-2 F300
G90

CameronLamont, 2018-08-22:
This may not be related ultimately - I not seeing the error but in a sense I'm getting no filament coming out at all if the layer and extrusion is small enough.  I'm trying to calibrate a 0.1mm nozzle so it may be PEBKAC on my part.

I was tracing through the python code and noted the max extrude cross section, I guess I'm wondering if this also an implicit minimum cross section too? (I can see the extruder is not moving at all so was wondering if I was actually seeing a rounding issue).

I am reading the code for the first time, not quite comprehending the intent yet but wanted to ask about if there are any data types in play that may have the effect of truncating a super small extruder amount ( a bit like the infinitesimally small xyz move mentioned in the #461 comments.

KevinOConnor, 2018-08-23:
@CameronLamont - there is no equivalent minimum cross section.  If you're having issues with the stepper, make sure you run through the steps in: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

KevinOConnor, 2018-08-23:
I'm going to close this issue in favor of the existing issue #461.

closed 2018-08-23 by KevinOConnor