Klipper3d/klipper #540: Filament runout sensing with TMC2130?

BKLronin, 2018-08-15:
I just wondered if we could use the SGT value and the diag pin as runout sensor. If we set sgt to somewhat low during extrusion so it triggers the diag pin all the time and we invert the logic we should be able to reliably detect a filament runout with the diag pin going to low as the load drops. Otherwise usable as jam detection but I personally find runout more important.

Haven´t found any previous discussions on that topic. Would be great as I´m refusing to integrate mechanical detection or lasers etc.

Arksine, 2018-08-15:
Unfortunately stallguard requires higher velocities (50mm/sec) to function properly, so it can't be used for filament runout/jam detection.

BKLronin, 2018-08-15:
A bummer but should have known this, sorry. Checked for the speeds, even at 120mm/s printing and 3:1 gearing the extruder runs only at 15mm/s ca.

Thanks

closed 2018-08-15 by BKLronin